stm32-rs/stm32h7xx-hal #1: Integrate to stm32-rs org

astraw, 2019-05-25:
@adamgreig @therealprof

Would it be possible to move this repository into the stm32-rs org? My plan would then be to add a repository link in Cargo.toml, release it (in its current unfinished state), and depend on the released version in https://github.com/astraw/nucleo-h743zi .

Also, PRs and suggestions are welcomed.

Thanks.

therealprof, 2019-05-25:
@astraw Absolutely! Please go ahead.

astraw, 2019-05-25:
Great. I think you have to make me a member of the stm32-rs org because I get "You don’t have the permission to create repositories on stm32-rs" when I try to transfer ownership.

adamgreig, 2019-05-26:
I've invited you to the stm32-rs organisation, welcome!

astraw, 2019-05-26:
Great, thanks. I transferred the repo now. I'll work on the next steps.

closed 2019-05-26 by astraw