stm32-rs/stm32h7xx-hal #3: Already existing stm32h7x3-hal project

hargoniX, 2019-06-13:
Hey guys, I'm the owner of https://github.com/hargoniX/stm32h7x3-hal which I've been working on for the past few months and I was wondering if you'd be interested in me pring all my implementations you don't have yet, like for example the already tested I2C and the SPI one  over to you and continue my work on your repository.

jordens, 2019-06-13:
IMO, please do merge the two.
There is #2 pending which will have a big impact as well so you'll have to sort that out.
Does it look like there are significant incompatibilities between what this would like after #2 and what merging your crate would look like?

hargoniX, 2019-06-13:
I would probably have to change my type state models for Pins in alternate function due to the gpio.rs changes but apart from that I don't see much of a change required no.

hargoniX, 2019-06-13:
@jordens How exactly would I do this type of development (doing a pull request which relies on code from another pull request but still compiles successfully on its own)

jordens, 2019-06-13:
Maybe landing one PR first and then revising the other or splitting both into incremental PRs and then interleaving them might be the only practical approach.

hargoniX, 2019-06-13:
Alright, I'm gonna wait until #2 is merged then and gonna make my code compatible with it in the mean time so I can shoot the PR out once it's ready.

astraw, 2019-06-13:
Great, thanks for filing this issue - in general I agree that merging is in principle the best thing to do.

closed 2019-06-13 by hargoniX