stm32-rs/stm32h7xx-hal #9: Tracking Issue for Revision V silicon

richardeoin, 2019-06-13:
ST has released a new silicon revision for the `H742/742/750/753` with significant changes.

* [eevblog](https://www.eevblog.com/forum/microcontrollers/stm32h7-series-revision-beware-of-the-changes!/)
* [ST community](https://community.st.com/s/question/0D50X0000Aixasy/stm32h7-cpu-speed-increase)

Please report or tag issues with Revision V here! Also see [upstream](https://github.com/stm32-rs/stm32-rs).

hargoniX, 2019-06-14:
In the eevblog link above I see two notes regarding the ADC (which as mentioned in #10 would've been my next addition)
```
-ADC Now has a extra 'divide by 2' on its clock inside it, making existing clock configurations too slow
-Changes to the ADC registers (For example a value in a register that used to set the ADC to 8bit mode is now setting it to 'reserved' so likely doesn't work)
```
Now I assume this could be easily resolved using feature gates, however a feature gate named "V" or "Revision-V" might be hard for the average user to understand as they are probably not aware of Revision V.

richardeoin, 2019-06-15:
Yes, these are significant changes! For now I say implement for Rev Y, then add `-rev_v` feature gates when someone has V silicon to test on.

jordens, 2019-06-17:
FYI: The NUCLEO-H743ZI**2** I just got from Digikey has Silicon rev V (and non-detachable ST-Link-v3)

jordens, 2019-07-11:
https://github.com/stm32-rs/stm32-rs/pull/247 for the PAC split.

jordens, 2019-07-29:
@richardeoin Can we close this? The crates/features are nicely split to address this, right?