stm32-rs/stm32h7xx-hal #34: Implementing peripherals which arent in embedded hal yet

hargoniX, 2019-07-31:
If somebody was to use this project as a basis for peripherals which aren't defined (and maybe never will be because they are too specific to this or the stm32 series) he would fail at the simple fact that he cannot use the RCC registers in the Ccdr at the moment. Is there any workaround to this which comes to mind (apart from obviously setting it from pub (crate) to normal pub))

richardeoin, 2019-08-03:
The consensus seems to be that we should only implement peripherals with `embedded-hal` traits here. This means that if `embedded-hal` does define traits for some other peripheral in the future, we don't have to remove our implementation before creating the `embedded-hal` version.

Support for other peripherals can come from other crates, but as you say they can't access `RCC`. The obvious answer for these crates is `unsafe`, like so:
```rust
        {
            // Unsafe: only access bits specific to DSIHOST
            let rcc = unsafe { &*stm32::RCC::ptr() };

            // Enable clock and reset
            rcc.apb3enr.modify(|_, w| w.dsien().enabled());
            rcc.apb3rstr.modify(|_, w| w.dsirst().set_bit());
            rcc.apb3rstr.modify(|_, w| w.dsirst().clear_bit());
        }
```
I think this is my preferred solution for sheer simplicity, and the scope of the `unsafe` is limited.

Alternatively we could implement setters/getters for each peripheral not used in this crate:
```rust
struct DsiHostEnable {
    _0: ()
}
impl DsiHostEnable {
    pub fn set() {
         ...
    }
    pub fn clear() {
         ...
    }
}
struct DsiHost {
    pub enr: DsiHostEnable,
    ...
}
```
But I think that these methods are excessive given that they're just calling `unsafe` under the hood anyhow. The only advantage is that they would be removed when this crate implements that peripheral. That makes the crate calling them safe against future versions of this crate.

hargoniX, 2019-08-03:
However things like DMA seem to be pretty common to be implemented even though they are not in embedded hal....but I assume that's because they're just a mean of internal data transfer.

closed 2019-10-03 by hargoniX