stm32-rs/stm32h7xx-hal #35: [Bug] ADC initialization might panic

Jan561, 2019-08-14:
### Description and Reproduction
Currently, the ADC initialization might panic if following happens:
- Watchdog enabled
- ADC initialized and enabled
- Panic occurs, causing watchdog to restart the software
- ADC initialized again (and thus calibrated) -> fails because:
  - Reset isn't being triggered because the clock of the ADC is still running (see adc.rs, line 283)
  - Therefore, ADC is still enabled
  -> ADC calibration will panic

### Possible fix

Upon initialization, check if ADC is enabled and disable it then.

richardeoin, 2019-08-14:
Thanks for detailed explanation!

## Possible Fix

Yes, this would work. Even better - toggle the reset line of every peripheral (excluding the CPU) on initialisation. This is good practice imo.

## API flaw

The ultimate reason for this is that when `Adc::adc1(dp.ADC1, ..)` is called, the HAL has to decide if the user meant to reset both ADC1 and ADC2, or neither. The hardware does not allow them to be reset separately. 

It currently tries to guess based on the enabled state, but in this use case it gets your intentions wrong.

The only option is to change the API to make the intention unambiguous. You could remove `Adc::adc1()` and `Adc::adc2()` and replace them with a `Adc::dual(dp.ADC1, dp.ADC2, ..)` function.

Jan561, 2019-08-14:
Yeah, good points. Currently I'm working on DMA, but when I finished that I can head over to the ADC.

richardeoin, 2020-07-27:
Closed by #94

closed 2020-07-27 by richardeoin