stm32-rs/stm32h7xx-hal #39: README on crates.io

hargoniX, 2019-10-14:
The README on crates.io doesn't seem to be up to date as of now, at least it still states that we require nightly / beta to build which isn't the case since the stabilization of 1.37 anymore, I think we should update that.

Secondly our build for docs.rs is currently failing, however I've already been pointed at a way to fix this and will look into that in a few minutes.

As a side not as to where this is coming from I'm currently doing a PR to get our crate into the awesome-embedded-rust list: https://github.com/rust-embedded/awesome-embedded-rust/pull/209

richardeoin, 2019-10-20:
v0.2.0 on crates.io is *exactly* what was reviewed in PR #33 (no rebases, no extra commits)

richardeoin, 2019-10-22:
I've committed and published a patch release v0.2.1 to fix this. 

Thanks for the [limit increase](https://github.com/rust-lang/docs.rs/issues/434), should mean that the the docs.rs build will run cleanly.

closed 2019-10-23 by hargoniX