stm32-rs/stm32h7xx-hal #52: Documentation build is failing.

david-boles, 2020-03-04:
Hello, hopefully this isn't a known issue somewhere else.
[https://docs.rs/crate/stm32h7xx-hal/0.3.0](https://docs.rs/crate/stm32h7xx-hal/0.3.0)

richardeoin, 2020-03-04:
Hey, welcome and thank you for the issue! This is a result of a known change in docs.rs, but should absolutely be linked or described here too.

This [issue](https://github.com/rust-lang/docs.rs/issues/563) is the best place to start. It looks like there isn't an ideal solution yet, but a PR with a workaround or partial solution for the next release would be very welcome.

It's also possible to run `cargo doc` locally
```
git clone https://github.com/stm32-rs/stm32h7xx-hal.git
cd stm32h7xx-hal
cargo doc --features rt,stm32h743
```

david-boles, 2020-03-06:
@richardeoin Reading through that, I don't know how I would implement a workaround (I'm effectively completely new to Rust which doesn't help) but if there's something I'm missing I'd be happy to try. Thanks for the local solution!

richardeoin, 2020-03-06:
No problem! I've tagged the v0.4.0 milestone so this gets reviewed before the next release

richardeoin, 2020-03-18:
Now fixed 🎉  More information here: https://blog.rust-lang.org/2020/03/15/docs-rs-opt-into-fewer-targets.html

commit 26c1321b (2020-03-18): Build documentation for thumbv7em-none-eabihf target only

Closes #52
closed 2020-03-18 by richardeoin