stm32-rs/stm32h7xx-hal #64: Support custom SPI configurations

ryan-summers, 2020-04-18:
The existing SPI configuration assumes a lot of default configuration values (e.g. 8 bit frames, 1 frame per transaction, no delay between CS and SCK assertions). 

I propose that we update the Spi constructors to accept a `spi::Config` structure instead of the `spi::Mode` parameter. This `Config` structure would contain all of the device-specific implementation. To construct the `Config`, a builder semantic can be used:

```
Let config = Spi::Config::new(Mode {polarity: Polarity::IdleHigh, phase: Phase::CaptureOnSecondTransition})
    .frame_size(8) // Configure 8-bit transactions
    .swap_miso_mosi() // Swap MOSI and MISO lines
    .cs_delay(100e-6) // 100uS delay from CS -> SCK
    .freeze();

let spi = spi::Spi1::new(dp.SPI1, (sck, miso, mosi), config, 10.mhz(), &clocks);
```

This allows us to initially provide a minimal configuration as to what users require and expand on it as needed without changing the API at all (e.g. backwards compatible).

I'd be happy to implement this and PR it, just wanted to get some feedback first.

### Background
I'm currently working on a [platform](https://github.com/quartiq/stabilizer) using SPI to talk to some ADCs and DACs where the channel is half-duplex. Some additional configurations that I need are:
1. MSSI support (programmable delay from CS assertion to transaction)
2. Configurable transaction length (16 bit frames)
3. Support for swapping MOSI/MISO lines

richardeoin, 2020-04-18:
Hello! This sounds very useful. I already have PR https://github.com/stm32-rs/stm32h7xx-hal/pull/61 for 16-bit frames that will be merged before v0.5, but would be keen to add even more functionality. That PR works well with the DAC124S085 for me, the PR also adds the documentation showing half-duplex usage.
```
let spi = dp.SPI1.spi((sck, spi::NoMiso, mosi), spi::MODE_0, 1.mhz(), &mut ccdr);
```

The general aim for this crate is to stick closely to the embedded-hal traits, to reduce the cognitive load for people switching between HALs (and allow device drivers to be interoperable). The [`FullDuplex`](https://docs.rs/embedded-hal/0.2.3/embedded_hal/spi/trait.FullDuplex.html) trait says that CS lines are out of scope, so I would not propose adding MSSI here. For the same reason, we should keep using the spi::MODE_n constants for configuration.

For swapping MOSI and MISO, the config struct is perfectly valid. But you could also implement it with an associated type on the [`Pins<SPI>`](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/spi.rs#L45) trait. Then you could simply write

```
let spi = dp.SPI1.spi((sck, miso, mosi), spi::MODE_0, 1.mhz(), &mut ccdr);
```

or

```
let spi = dp.SPI1.spi((sck, mosi, miso), spi::MODE_0, 1.mhz(), &mut ccdr);
```

and the correct choice would be made at compile time.

ryan-summers, 2020-04-18:
From a higher level perspective, how should we manage implementing device-specific implementations (such as MSSI)? Currently, because the HAL takes ownership of the register block, there's no way to manipulate the registers without unsafe code. Would we rather just expose the register block as a public member of the Spi object?

richardeoin, 2020-04-18:
If you're writing a device driver to operate across all embedded-hal targets (stm32, nRF, rasberry pi and so on), then you have to compose the embedded hal traits only. For the an SPI DAC that needs MSSI, that works out as

```rust
                self.nsync.set_high()?; // Deassert

                // Wait tSYNC = 10ns
                delay_source.delay_us(1);

                self.nsync.set_low()?; // Assert

                // Command
                block!(self.spi.send(command))
                    .map_err(|e| TIDacError::SPIError(e))?;
                // Wait for it on the wire
                let _ = block!(self.spi.read())
                    .map_err(|e| TIDacError::SPIError(e))?;

                self.nsync.set_high()?; // Deassert

                delay_source.delay_us(1);

                self.nsync.set_low()?; // Idle Low
```

Which is really inefficient, but is portable. To get a more powerful implementation you'd likely need to propose adding another trait in embedded-hal

ryan-summers, 2020-04-19:
Let me try to phrase a hypothetical:

Suppose a user wants to use a SPI implementation for a specific platform, where they know the hardware will not change and are not interested in making a platform-agnostic driver (e.g. an optimized use case). For efficiency reasons, the user wants to specify transaction endianness, transaction bit-sizes, and automatic CRC validation within the SPI peripheral so they don't have to take a run-time overhead.

With the current implementation, the only two options available to the user are (to my knowledge):
1. Do not use this hal for the `SPI` implementation and instead roll their own. In this case, they would essentially duplicate a lot of code from the HAL, since the read/write/configure steps are nearly identical with a few different bits.
2. Create another reference to the SPI configuration registers in an `unsafe` manner to manually modify the registers that they are interested in.

Essentially, if we don't give this user access (in some way or another) to the configuration registers that they need, we preclude their usage of this hal in a safe manner.

### Additional note (Opinion piece)
My interpretation of the `embedded-hal` is that it provides a platform-agnostic interface for device drivers, _not_ that it provides platform agnostic configuration. Once the SPI is configured (in a platform-dependant manner), it can still operate in a platform-agnostic means (e.g. through the `embedded-hal` traits). Although reducing cognitive load on users is generally good and should be supported, I don't think that we want to restrict functionality of the SPI driver because of it (but this is entirely my own opinion).

richardeoin, 2020-04-19:
Agreed that neither of those two options is particularly attractive, and therefore on balance adding non-embedded-hal functionality is OK as long as we don't interfere with the embedded-hal "happy-path". I didn't really think about that possibility in previous replies.

I'm imagining something like this, but feel free to change to what you were thinking.

```rust
fn spi_temperautre_sensor(spi: impl embedded_hal::spi::FullDuplex) {
    spi.send(99_u8).ok();
}

#[entry]
fn main() -> ! {
    ...
    let spi1 = dp.SPI1.spi((sck, miso, mosi), spi::MODE_0, 1.mhz(), &mut ccdr);

    spi_temperautre_sensor(spi1); // Works

    let config = ....;
    let spi2: SpiWithCS<...> = dp.SPI2.spi((sck, miso, mosi, cs), config, 1.mhz(), &mut ccdr);

    spi_temperautre_sensor(spi2); // Either compile error (trait not implemented),
    // or produces a transaction where the CS pin is unchanged

    spi2.send_with_cs().ok(); // Transaction with CS pin
}
```

Agreed embedded-hal doesn't specify configuration (Otherwise we couldn't invent `Ccdr` to deal with the H7's super complicated clock distribution!). So I'm ok with changing the configuration API if needed as this crate is still quite young.

In short we should try try to provide as much functionality to `FullDuplex` trait users as possible (the miso/mosi swap configuration would be cool), but for functionality that doesn't fit in `FullDuplex` (like CS pin and MSSI), we should implement it outside `FullDuplex` and give users the option to ignore it for the configuration (the cognitive load bit).

ryan-summers, 2020-04-19:
I'd be happy to look into an implementation for this soon (the swapped SPI pins would definitely be worth adding to `FullDuplex`)

ryan-summers, 2020-04-19:
> For swapping MOSI and MISO, the config struct is perfectly valid. But you could also implement it with an associated type on the [...]

My only concern with this is that the user isn't explicitly saying that MISO and MOSI are swapped, and i would be quite easy for someone to accidentally write (sck, mosi, miso) instead of (sck, miso, mosi) and not understand why it wasn't working.

richardeoin, 2020-04-19:
Yes, agreed that is a possible pitfall. It's already a pain as you have to explicitly look at the HAL docs to see that the correct order is indeed (sck, miso, mosi).

One solution is a marker type
```
(sck, mosi, miso, Swap)
```
Or just include it as part of the config

richardeoin, 2020-07-27:
Closed by #68

closed 2020-07-27 by richardeoin