stm32-rs/stm32h7xx-hal #66: Expose RCC registers for external usage

ryan-summers, 2020-04-18:
When the HAL is used in an application, there are (many) peripherals that are not yet supported by the HAL (such as ethernet, quadspi, DMA, etc.). Often, these peripherals may need a custom implementation for performance or specific use-case considerations. 

The `Rcc::Ccdr` should expose the underlying RCC register block publicly so that other drivers may modify the necessary RCC registers to enable clocks and reset peripherals as needed. This is currently done only for the HAL crate, but it may be appropriate to allow this block to be exposed for all usage. 

As currently implemented, the HAL takes exclusive ownership of the RCC registers, so it's impossible to implement application-specific drivers outside of the HAL alongside drivers that utilize the HAL implementation.

richardeoin, 2020-04-18:
Indeed, many peripherals don't have embedded-hal traits and hence don't really belong here. The DMA is probably one exception because it tightly integrates with other peripherals, and there is a WIP PR for that in https://github.com/stm32-rs/stm32h7xx-hal/pull/48

Others work well as external crates, like [stm32h7-ethernet](https://crates.io/crates/stm32h7-ethernet) (Although @jordens et al's implementation in stabilizer looks really good too!).

External crates can already modify the RCC registers as needed with `unsafe`, stm32h7-ethernet does that [here](https://docs.rs/stm32h7-ethernet/0.1.1/src/stm32h7_ethernet/ethernet.rs.html#419).

ryan-summers, 2020-04-18:
Although we can already access the registers using `unsafe` as mentioned, is that how we would like external drivers to access the register blocks? It would bypass the ownership concept of the register block, but it seems fine for the use case of the RCC. The main point takeaway I have is that external drivers may want the `Ccdr` structure anyways to access the configured clock frequencies and the register block is already contained in there, so it seems a bit odd to have to unsafely re-instantiate a pointer to the registers just to modify them.

richardeoin, 2020-04-18:
We really don't want external crates to access the RCC register block, as they could do anything! (change bus clocks, reset peripherals, even reset the core). Unfortunately with the current PAC we can't stop external crates doing these things with unsafe. (There are some [ideas](https://github.com/japaric/embedded2020/tree/master/host/regen#singletons-are-always-singletons) about changing this, but not likely any time soon).

Ideally we would `.split()` the RCC register block into many pieces with a large number of safe functions on each. Then the appropriate parts would be passed to this HAL, and other parts to other crates. The beginnings of this [exist already](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/rcc/mod.rs#L164), but it's nowhere near complete. So the reference to the register block is contained in the `Ccdr` structure as a hack until that's done.

ryan-summers, 2020-04-18:
The current implementation you link to above _does_ expose the respective AHB bus registers in a public API, but the actual register accesses are public only to the `hal`crate currently (e.g. `ccdr.ahb4.enr().modify(|_, w| ...)` can only be completed within this crate). Is the intent that the bus registers would also be public then?

richardeoin, 2020-04-18:
I think the intention was the a completely safe implementation would be built, which could then be public.

Something like:

```rust
impl AHB1 {
    pub fn split(self) -> (RccEth1Mac, RccAdc12, RccDma2, RccDma1) {
        (RccEth1Mac { _0: () }, ...)
    }
}
struct RccEthMac1 {
    _0: ()
}
impl RccEthMac1 {
    pub fn enable(&mut self) {
        // Only change bit 15 of AHB1ENR
    }
    pub fn reset(&mut self) { ... }
}
...
```

commit 746e8021 (2020-06-25): v0.6.0

See CHANGELOG

Closes #66
closed 2020-06-25 by richardeoin