stm32-rs/stm32h7xx-hal #80: DMA API

hargoniX, 2020-05-26:
During today's embedded WG meeting James mentioned that:
> If you see anyone working on DMA stuff, ping them and see if they could move their work over to the new APIs :)
Referring to these https://github.com/ra-kete/dma-poc APIs.

So @jan561 should maybe try and look into directing his efforts towards getting DMA running towards these APIs, if enough succesul implementations show up they might be added to embedded-hal soon :tm: which would be a big plus embedded Rust.

mtthw-meyer, 2020-07-14:
I except I'll end up with an implementation of DMA for #90.

hargoniX, 2020-08-25:
Since embedded-dma has been taken over by the embedded wg the new way to go is going to be to implement that!

mtthw-meyer, 2020-08-25:
Ok. Is there a issue I can track for that?

richardeoin, 2020-08-25:
Yup! This is led by the working group, so you need to look there https://github.com/rust-embedded/wg/issues/486

ryan-summers, 2020-10-14:
`embedded-dma` is now officially supported by the HAL team, so I think it would be safe to use. I have a project that's looking to use memory -> peripheral for QSPI transactions, so I may have some time to devote to this.

mtthw-meyer, 2020-10-14:
Nice. Having a working use case would make it easier to implement for SAI.

richardeoin, 2020-10-14:
Agreed! A small [dma module](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/dma.rs) was merged as part of #99, but I think that can just be overwritten with an `embedded-dma` implementation.

mtthw-meyer, 2020-10-14:
Yeah, that was me taking a stab at getting some basic DMA stuff in. I was originally going to support DMA in the SAI implementation but it proved to be too much at once.

antoinevg, 2020-10-18:
I also mentioned this on #153 but for reference here's a link to my proof-of-concept example using the SAI peripheral with DMA:

https://github.com/antoinevg/stm32h7xx-hal/blob/sai-dma/examples/sai-dma-passthru.rs

richardeoin, 2021-02-14:
Closed by #153

closed 2021-02-14 by richardeoin