stm32-rs/stm32h7xx-hal #87: Unused trait in Serial?

hargoniX, 2020-06-20:
So I was just reading through our UART/USART implementation in serial.rs and there we define the trait `pub trait PinCk<USART> {}` in line 155. It's only ever used again in the macro that implements it en mass for all the pins that are valid clock pins for USART. However I believe the correct intention should be to implement the Pin trait for a (TX, RX, CK) triple for all USART constructors right? Otherwise the trait doesn't bring any additional safety guarantees as far as I can see (feel free to correct me on that).

hargoniX, 2020-06-20:
Pinging @richardeoin as he implemented it

richardeoin, 2020-06-20:
You're right, the `PinCk` trait doesn't do anything here. It's really there for future use, in case we implement one of the synchronous modes in the HAL.

It's also publicly available as a trait bound, which is possibly useful in other crates.