stm32-rs/stm32h7xx-hal #88: GPIOC Unknown Field

mtthw-meyer, 2020-06-23:
I can compile this without any issues 
cargo build --example blinky --features=stm32h750v,rt

But if I take the same code, paste it into a new crate, copying the .cargo/config and dependencies it won't compile. I've compared everything I can think of and there is just something missing that I can't figure out.

```
cargo build
   Compiling daisy-lib2 v0.1.0 (/home/zerikin/rust/daisy-lib2)
error[E0609]: no field `GPIOC` on type `stm32h7xx_hal::rcc::rec::PeripheralREC`
  --> src/main.rs:28:48
   |
28 |     let gpioc = dp.GPIOC.split(ccdr.peripheral.GPIOC);
   |                                                ^^^^^ unknown field
   |
   = note: available fields are: `ETH1MAC`, `DMA2`, `DMA1`, `SDMMC2`, `HASH` ... and 22 others

error: aborting due to previous error

For more information about this error, try `rustc --explain E0609`.
error: could not compile `daisy-lib2`.

To learn more, run the command again with --verbose.
```

richardeoin, 2020-06-23:
Hey @mtthw-meyer, thanks for reporting!

Just a quick note - The latest version on github has some breaking API changes that aren't released or documented yet. The current version on crates.io is v0.5.0, and you can find the examples for that on the [v0.5.0 tag](https://github.com/stm32-rs/stm32h7xx-hal/releases/tag/v0.5.0)

mtthw-meyer, 2020-06-23:
That would do it.

closed 2020-06-23 by mtthw-meyer

richardeoin, 2020-06-24:
Great! I'd be open to suggestions for improving user experience issues like this one. Particularly as the crate matures and more people start using it, it makes sense to be more careful.

Ps. The electrosmith daisy looks like a gorgeous board, and even better with Rust! Hopefully my [fmc](https://github.com/richardeoin/stm32h7-fmc) and [sdmmc](https://github.com/richardeoin/stm32h7-sdmmc) crates will be useful too. Don't hesitate to ask on the rust-embedded matrix channel if you want advice / help

mtthw-meyer, 2020-06-24:
Trying to get past the initial hurdle of getting Rust to run on the board at all still. I've gotten onto the matrix board and just got SWD up so hopefully I can figure out where it's going wrong.

richardeoin, 2020-06-24:
Yes that part can be tricky, depending on the debugger you have. Those kind of problems are also on-topic for the matrix channel, so that's the best place to ask. Good luck!