stm32-rs/stm32h7xx-hal #93: Prec structures preclude using ADC1 and ADC2 simultaneously

ryan-summers, 2020-07-04:


ryan-summers, 2020-07-04:
@richardeoin Perhaps we could update the ADC constructor to just borrow the ADC PREC structure, which would allow it to be used more than once? Curious to hear your thoughts. We could potentially also generate duplicate PREC structures for ADC1 and ADC2, but I feel like that would obfuscate the fact that they're essentially sharing resources.

richardeoin, 2020-07-04:
I think this is the only case where a single reset line is used to control two peripherals that have separate definitions in the SVD.

We could fix this the in ADC HAL by initialising both ADC1 and ADC2 at the same time. The usage would then look like:

```
Adc::adc12(dp.ADC1, dp.ADC2, ccdr..., ...) -> (adc1, adc2)
```

richardeoin, 2020-07-04:
So if you were just using ADC1:

```
let (adc1, _) = Adc::adc12(dp.ADC1, dp.ADC2, ccdr..., ...);

ryan-summers, 2020-07-04:
I like that concept as it makes it very clear to the user that the peripherals are intertwined.

richardeoin, 2020-07-04:
Indeed, it forces any alternative ADC HAL implementations to acknowledge that ADC1 and ADC2 are linked

ryan-summers, 2020-07-04:
I'd be happy to write up a PR for this change. It'd be nice to get some of the `stabilizer` code base as back-on-track with the HAL as possible.

richardeoin, 2020-07-04:
I'm most of the way through making the change, will push a PR shortly. I'll open the branch to collaborators though, if there's anything you want to add.

ryan-summers, 2020-07-23:
Fixed in https://github.com/stm32-rs/stm32h7xx-hal/pull/94

closed 2020-07-23 by ryan-summers