stm32-rs/stm32h7xx-hal #95: ST have renamed D3 to SRD on new A3/B3 parts

adamgreig, 2020-07-11:
Hi all,

This issue really affects the PACs but I expect it will have an impact on the HAL, and the HAL developers are probably in a good position to comment on it, so I thought I'd ask here first.

It's been pointed out that ST have started calling "SmartRun", "SmartRun Domain", and "SRD" what used to be called "D3". Our SVDs for the A3/B3/B0 parts use "D3", for example in `PWR:CPUCR:RUN_D3` which is now called `RUN_SRD` in the reference manual. It seems like the older H7 parts are not renamed (yet?).

What do you think we should do about it? Options seemingly include:

* Do nothing, keep D3, which maintains easier compatibility with other H7 parts for clock and memory setup, but disagrees with the reference manual
* Rename D3 to SRD in the SVDs, so HAL and user code will have to handle these devices separately
* Other?

I assume future H7 devices will use the new names, which might suggest we stick to the reference manuals and at some point have to rename the current SVDs.

hargoniX, 2020-07-11:
I think disagreeing with the Reference Manual is usually a bad idea, especially naming wise. Imagine if someone wanted to look into the HAL for a potential bug fix but they find terminology regarding registers that is not mentioned in the reference manual, that doesn't seem like a particularly good idea for people who aren't familiar with the naming issue.

richardeoin, 2020-07-11:
Yup, I'm ok with renaming the SVDs and cleaning up the difference here in feature gates

richardeoin, 2020-10-14:
Closed by #122

closed 2020-10-14 by richardeoin