stm32-rs/stm32h7xx-hal #117: ResetEnable trait methods require ownership

richardeoin, 2020-08-04:
The [`ResetEnable` trait](https://docs.rs/stm32h7xx-hal/0.6.0/stm32h7xx_hal/rcc/rec/trait.ResetEnable.html) is implemented for Peripheral Reset Enable Control (PREC) [ZSTs](https://doc.rust-lang.org/nomicon/exotic-sizes.html#zero-sized-types-zsts). However, its methods require `self`, so it cannot be used when we only have a `&mut-` reference to the PREC.

Proposal: 

Change the `ResetEnable` trait to implement these methods for `&mut self` instead.

Advantages:

* Usability, see https://github.com/stm32-rs/stm32h7xx-hal/pull/116

Downsides:

* Allows the PREC to be `reset`/`enabled` when we only hold a `&mut-` reference to it. Currently if the PREC is part of a struct, we can be sure the peripheral will not be reset / disabled until that struct's destructor.

hargoniX, 2020-08-04:
I would not feel good about dropping this guarantee for all peripherals just because of FMC.

having the ResetEnable trait provide an  unsafe method that takes &mut and internally makes a self + explicitly discouraging use of that method without a very good reason seems like a better solution for this.

While the guarantee is of course not provided anymore for FMC then, we would at least ensure no developer in the future accidentally invalidates the guarantee while in reality we could still provide it if the API was (self) instead of (&mut self)

Alternatively we could have a fully separate trait (since this will be done on a peripheral to peripheral difference) like UnsafeResetEnable which has the same method for &mut self. This would have the added advantage, that in docs.rs the user can see whether the peripheral is ResetEnable (i.e. has the destructor guarantee) or UnsafeResetEnable (i.e. does not have the guarantee) -> greater transparency for the consumer of the HAL regarding the guarantees it provides + the difference is obvious inside the type system already

Personally I'd favour the separate trait approach, what I really would not like is that we just drop this nice guarantee for all peripherals just because of one black sheep.

mlamoore, 2020-08-04:
I agree; a separate trait like UnsafeResetEnable seems much safer and more transparent.

The ResetEnable trait could even have an into_shared or into_mut method that consumes self and provides an UnsafeResetEnable equivalent, so it's easy to work with perpetual libraries that require either one.

ryan-summers, 2020-08-05:
@hargoniX 
> I would not feel good about dropping this guarantee for all peripherals just because of FMC.

When you refer to "this guarantee", I'm assuming you are referring to 
> * Downsides:
>
> Allows the PREC to be reset/enabled when we only hold a &mut- reference to it. Currently if the PREC is part of a struct, we can be sure the peripheral will not be reset / disabled until that struct's destructor.

Is the intent that the the PREC explicitly **cannot** be used when a member of a struct? E.g. the PREC _must_ be moved into some peripheral before it can be used to reset/enable a peripheral? That seems sound to me, and the FMC can actually employ special `Option` wrapping of the PREC to workaround the ownership issue - I don't know if we need a fully separate trait just for a single use case yet.

hargoniX, 2020-08-05:
I'd imagine that if the FMC requires us to do this with an &mut API instead of our current one there might be other peripherals showing up in the future that would require us to do it like that, I'd just prefer to be prepared in that case already.