stm32-rs/stm32h7xx-hal #123: UART bug

hargoniX, 2020-08-11:
In a mail by @eeboo he brought up the following repository: https://github.com/eeboo/stm32h743zi-test/

In there you will find two examples, one tagged ok, the other tagged ko. The ko example is producing garbage on his logic analyzer.
As one can see in his example code the only difference is:
```rs
let tx = gpiod.pd8.into_alternate_af7();
let rx = gpiod.pd9.into_alternate_af7();
```
in the Ok example, vs

```rs
let tx = gpiob.pb9.into_alternate_af8();
let rx = gpiob.pb8.into_alternate_af8();
```
in the not working example. (and only now am I realizing that in my response I actually misread, its not pd8 being used for both but pb8 for the not working one)

We should definitely investigate on this.

hargoniX, 2020-08-11:
I can reproduce that Ok works
However Ko works perfectly fine for me as well, at least when modifying it so it echo's back to itself and then prints what it receives via RTT.

So this is an issue specific to @eeboo 's setup since I have a stm32h743zi chip on my end as well.
Would someone have an idea what we could do to figure out what's wrong with his setup?

richardeoin, 2020-08-11:
Thanks for reporting this, it's good to have it in the issue tracker too.

Just a note that I also tried the UART4 peripheral in the past and it worked for me. But I was using pins PA11 and PA12.

Could you try using those pins instead? Are they accessible on the Nucleo board? It could actually be an electrical issue (broken trace or pad perhaps)

eeboo, 2020-08-11:
Hello thanks for your comments.
To avoid electrical issue I just run a new test generated with cubemx with UART 4 and PB9 as TX
It's working perfectly fine.

[image]

https://github.com/eeboo/st-nucleo-h743ZI-cubemx-serial-test

eeboo, 2020-08-11:
Hello my setup caused the issue.

when the logic analyzer is not plugged to the same computer as the nucleo board the interpreted data is corrupted.
when it is plugged on the same computer data is ok.

thank you very much for your feedback and sorry for wasting your time

I think you should close this issue

closed 2020-08-11 by hargoniX

richardeoin, 2020-08-11:
Ok, great that you've found the problem. Please make new issues if you find other problems with the HAL 😃 

I did test the `serial_usart4_ko` example on a stm32h743vi rev Y, it worked for me also.