stm32-rs/stm32h7xx-hal #145: SDMMC: Cannot init_card() after card is removed and inserted

mattico, 2020-09-30:
Using the code at https://github.com/mattico/miscomp-repro extracted and simplified from my firmware.

Trying to run `init_card()` a second time after the card is removed and re-inserted always returns an error here: https://github.com/stm32-rs/stm32h7xx-hal/blob/e6da7dc7bf9cd9f85e4f84a980a94e8cbf18be34/src/sdmmc.rs#L520

Since there's no way to `free()` and re-initialize the SDMMC it seems like this should work?

mattico, 2020-09-30:
Actually the error doesn't happen at sdmmc.rs:520, that was a debuginfo bug? It seems that `get_scr` returns `Error::Timeout`.

mattico, 2021-04-26:
More precisely: the card can be inserted at runtime but it only works the first time it's inserted.

Looking at **55.6.7 Reset and card cycle power** in the RM, there are a number of steps that need to occur to reset the card, which includes completely resetting the SDMMC peripheral (the PWRCTRL bits can't be written after being set to ON). Perhaps the card is getting spurious data when inserted due to the peripheral still being enabled after the first insertion.

olback, 2022-02-04:
@mattico Did you ever get this working?

mattico, 2022-02-04:
Nope!