stm32-rs/stm32h7xx-hal #148: Implement tick_timer() for lptimers

richardeoin, 2020-10-05:
Ref https://github.com/stm32-rs/stm32h7xx-hal/pull/114

The `tick_timer` method is not implemented for LPTIM[1-5]

*Is there some limitation of the LPTIMs that makes it difficult?*

The LPTIMs require a specific initialisation sequence (for example, CFGR must be set before enabling the timer, ARR must be set after), so it's not difficult but rather a little tedious to check against the RM that it's correct.

noppej, 2020-12-03:
@richardeoin 
I needed the `tick_timer` method, so I have a push available with this code.   There are a couple of differences in how LPTIM'ers work relative to the other TIM'ers, but nothing super complex. 

My code includes a couple of (proposed) fixes such as 
- fn counter() sometimes gave incomprehensible results, so I changed the  
`self.tim.cnt.read()bits()`  
to  
`self.tim.cnt.read().cnt().bits().into()`

richardeoin, 2020-12-03:
Hi @noppej That's great, proposed improvements and fixes are very welcome! Normally we do this by Pull Requests, so if you make a Pull Request with your code I would review it.

noppej, 2020-12-05:
Hi @richardeoin ... I will gladly create a pull request. Before I do, I'd like to get your view on something, so that I can fix that at the same time. 

Currently, for regular (non low power) timers, the `fn set_tick_freq<T>(&mut self, frequency: T)`, sets a reload value in the ARR register. This function, is also called from `fn tick_timer<T>(self, frequency: T, prec: Self::Rec, clocks: &CoreClocks)` .  
Is it just me, or should the tick_timer set the frequency (PSC) ONLY, and NOT set the ARR?

richardeoin, 2020-12-08:
I think `set_tick_freq` does set the ARR to 0xFFFF for 16-bit timers, and 0xFFFF_FFFF for 32-bit timers. That's those two lines https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/timer.rs#L376-L377

The documentation on the `tick_timer` method in its trait definition says that it "Counts from 0 to the counter's maximum value". That sentence could be made more precise (does it include the maximum value or not?), but setting ARR to the maximum value seem reasonable based on that.