stm32-rs/stm32h7xx-hal #157: DMA module does not reset/enable during initialization

ryan-summers, 2020-10-28:
When `DMA::split()` is called, the PREC is not provided and the DMA does not set the reset/enable bits in the RCC.

Further configuration and usage of the DMA is then inoperative because the peripheral is not enabled.

richardeoin, 2020-10-28:
Fixed in #153 . The DMA API in the current master and release 0.8.0 is really only intended to live for one release cycle, and you should have got a depreciation warning on the current master or v0.8.0.

I'd be interested in your feedback on #153 @ryan-summers . For the QSPI it's possible ST's intention was that it would be used with the MDMA, as there's two connections from the QSPI to the hardware request lines for the MDMA. But it might be possible to use the ordinary DMA1/2 also. I did some [background](https://github.com/stm32-rs/svdtools/pull/42) [work](https://github.com/stm32-rs/stm32-rs/pull/454) for the [MDMA](https://github.com/stm32-rs/stm32h7xx-hal/pull/153/files#diff-83bb17ad20b71ad74e5fa57169f61d11aa6d64a5a06cee94537a70a3afebbc4bR163), but I've actually tried it out yet.

ryan-summers, 2020-10-29:
Yeah, I was working of the `master` branch and it looks like I pulled a version before the deprecation warning was added. I recently updated and see it now. I'll definitely take a look at #153 - we have some strange DMA use-case requirements for https://github.com/quartiq/stabilizer/ and I'm curious to see if there's a logical way to support them in the HAL (if not, I can figure out a work-around)

richardeoin, 2021-02-14:
Closed now that #153 is merged

closed 2021-02-14 by richardeoin