stm32-rs/stm32h7xx-hal #165: No QSPI support for memories

richardeoin, 2020-11-03:
Some of the features that would be required to communicate with memories over QSPI (instructions, dummy cycles..) are not supported

x37v, 2021-06-22:
I'm wanting to interface with 25LP064 QSPI flash memory and so I'm hoping my efforts can help this HAL.
I'm wondering if you all have any advice for the API for this?
Would it make sense to add `instruction: Option<u8>` to read and write and `mode: Option<u8>` to read (I haven't seen mode in write)? 
I'm assuming I can update the `imode` value based on the value set in `admode` ?

richardeoin, 2021-06-22:
I've started implementing some new methods for interfacing with OCTOSPI flash memories, and these are probably applicable to QSPI also. I'll make a PR with the progress so far..

richardeoin, 2021-06-22:
In the example in that PR there's an instruction with no address and no data phases ([line 80](https://github.com/stm32-rs/stm32h7xx-hal/pull/230/files#diff-37e1cf6c7c3da03fb936b858738b81830c681cced7575695592991fb5186d7ceR80)), as well as an instruction with both address and data phases ([line 84](https://github.com/stm32-rs/stm32h7xx-hal/pull/230/files#diff-37e1cf6c7c3da03fb936b858738b81830c681cced7575695592991fb5186d7ceR84)).

It should be close to working for QSPI memories also, but I've not tried it. If there's only small changes needed to work with the 25LP064, that would be very useful to know.

x37v, 2021-06-22:
> In the example in that PR there's an instruction with no address and no data phases ([line [...]

Great @richardeoin I'll give this a go soon.

Longer term, I'm guessing the original read/write would need to be altered to correctly do the `setup_extended`.. if they're not removed entirely?