stm32-rs/stm32h7xx-hal #169: Alternative Delay implementation

glindstedt, 2020-11-14:
If I want to use scheduling in `rtic` the init function does not have access to the `SYST` timer as described here (end paragraph of first section): [https://rtic.rs/0.5/book/en/by-example/timer-queue.html](https://rtic.rs/0.5/book/en/by-example/timer-queue.html).

This means that there's no way to construct the current delay implementation which requires the `SYST` timer.

[This comment](https://github.com/rtic-rs/cortex-m-rtic/issues/117#issuecomment-692751736) on the issue over at cortex-m-rtic suggest that HAL crates provide alternate Delay implementations to support this use case, potentially using `cortex_m::asm::delay`.

I tested this a bit locally but a delay based on `cortex_m::asm::delay` seemed pretty unstable and delayed for less time than I had calculated. Later I found this issue seem to explain the problem: [https://github.com/rust-embedded/cortex-m/issues/236](https://github.com/rust-embedded/cortex-m/issues/236)
The gist of it being that the current implementation is only accurate on M3/M4.

As far as I understand though, `Delay` in this crate is only really used for startup configuration where one has to wait _at least_ some amount of time, so it could still be useful to have an implementation based on `cortex_m::asm::delay` if it's augmented to account for the inaccuracy of the current implementation, and properly documented to inform the user.

If this seems reasonable I could provide a PR for adding this.

richardeoin, 2020-11-14:
Thanks for the thorough summary of the problem! I think it would be reasonable to offer alternative delay implementations in the HAL, if only because it's a small piece of functionality to require an extra crate for.

I have this snippet for those times when I need multiple objects that implement `DelayMs`/`DelayUs` https://gist.github.com/richardeoin/ba57bbc0618676100bbffe2ae28d05bc  Basing the implementation on a hardware timer avoids worrying about the core implementation. Feel free to use that snippet in a PR if you think it would be useful.

glindstedt, 2020-11-15:
The gist you linked was excellent, so I just used that without modification. It's much better than relying on the `cortex_m::asm::delay` function. I only had to change a few methods to take trait implementors instead of the `Delay` struct to make it work.

commit f3e10a1e (2021-01-10): Merge #170

170: Add DelayFromCountDownTimer for when SYST is unavailable r=richardeoin a=glindstedt

Fixes #169
PR #170 (2021-01-10): Add DelayFromCountDownTimer for when SYST is unavailable
Fixes #169
closed 2021-01-10 by bors[bot]