stm32-rs/stm32h7xx-hal #174: Improved timer implementation

ryan-summers, 2020-12-16:
The existing timer implementation provides itself well to giving the user a generic `Countdown` or `Periodic` object that can be used for regularly completing tasks, but there is limited control provided to the end-user.

For precision-control interfaces, timers may need to have a known tick rate (prescaler) and period (ARR). This can then be used for e.g. timer capture/compare channels and precise control of the timer peripheral.

I would propose that we extend the existing API for timers to allow for a new interface for construction / modification:
```
timer::set_tick_rate_and_period(&mut self, tick_rate: impl Into<Hertz>, period: u32);
```

Additionally, I would also propose we expand timers to include capture/compare channels as well. This way, capture/compare channels can be used to trigger DMA transfers (example available at https://github.com/quartiq/stabilizer/pull/195/files?file-filters%5B%5D=.lock&file-filters%5B%5D=.rs&file-filters%5B%5D=.toml&hide-deleted-files=true#diff-9cc696374e747d2ed0b0eee3bb74ff5ebe97de41fa24fbe30c1c9dc48cb50939)

CC: @noppej

This was brought up in https://github.com/stm32-rs/stm32h7xx-hal/pull/172

mlamoore, 2021-03-20:
I'm in favor of those changes.

If we're including timer and capture/compare functionality and integrating with DMA, why not merge it with PWM?

PWM already implements all the GPIO stuff we'd need for input capture. Additionally, I've run across situations in the past where I want to use one timer with one channel of input capture and another channel of PWM. The new advanced PWM PR I just pushed already includes ways to explicitly set prescaler and period instead of just frequency.

I could see something like this:

let (c1, c2, c3) = dp.TIM1.timer(...);

let mut c1 = c1.into_pwm();
let mut c2 = c2.into_timer();

That said, if we wanted to do that, the timer implementation would have to work with a free-running timer, grabbing the current count and doing the math to figure out when it should expire.

I'm interested in input capture (and encoder mode) anyways; I was planning to add triggered injected ADC sequences first, but I could work on this (at my very slow rate of not much free time).

everdrone, 2024-12-31:
Would love to see input capture mode implemented in the HAL. Has any further work been done on a separate branch?