stm32-rs/stm32h7xx-hal #180: Intended structure query

ostenning, 2021-01-12:
Hello, I'm trying to get a grasp of the paradigms of this library. I know this is probably a trivial question, but I'm not sure how to tackle it.

Essentially I have this:

```
fn main() -> ! {
   ...
   let gpiob = dp.GPIOB.split(ccdr.peripheral.GPIOB);
   
   // pass ccdr.clocks & ccdr.peripheral because we are unable to pass ccdr as its been moved for gpiob init
   drivers::oled::init(
       dp.TIM2,
       &ccdr.clocks,
       &ccdr.peripheral,
       gpiob.pb6,
       gpiob.pb7,
       dp.I2C1,
   );
}
```

But because `peripheralREC` is defined within a macro, I cannot declare this as an input parameter type for my `oled::init` function.

If `peripheralREC` cannot be passed, then it requires me to restructure the initialization method where I would initialize the `I2C` interface (which is what drives the oled display) in the main function and then pass this to `oled::init` which kind of breaks the encapsulation that I was intending for this.

What am I missing?

Kind regards

mattico, 2021-01-12:
The types you want to use for REC arguments are here: https://docs.rs/stm32h7xx-hal/0.8.0/stm32h7xx_hal/rcc/rec/index.html#structs

I encapsulate functionality like this:

```Rust
use stm32h7xx_hal::{
    gpio::*,
    hal::digital::v2::OutputPin,
    nb, pac,
    prelude::*,
    rcc::{self, rec},
    serial,
};

pub struct Gps {
    _tx: serial::Tx<pac::UART7>,
    rx: serial::Rx<pac::UART7>,
    reset: gpiog::PG1<Output<OpenDrain>>,
    _int: gpiog::PG0<Input<Floating>>,
    buffer: ArrayVec<[u8; 128]>,
}

impl Gps {
    pub fn new(
        tx: gpioe::PE8<Analog>,
        rx: gpioe::PE7<Analog>,
        uart: pac::UART7,
        uart_rec: rec::Uart7,
        reset: gpiog::PG1<Analog>,
        int: gpiog::PG0<Analog>,
        clocks: &rcc::CoreClocks,
    ) -> Self {
        let tx = tx.into_alternate_af7();
        let rx = rx.into_alternate_af7();
        let mut uart = uart.serial((tx, rx), 9600.bps(), uart_rec, clocks).unwrap();
        uart.listen(serial::Event::Rxne);
        let (_tx, rx) = uart.split();

        let mut reset = reset.into_open_drain_output();
        reset.set_high().unwrap();
        let _int = int.into_floating_input();

        Self {
            _tx,
            rx,
            reset,
            _int,
            buffer: ArrayVec::new(),
        }
    }
```

And then create like this:

```Rust
        let clocks = ccdr.clocks;
        let prec = ccdr.peripheral;

        let mut gps = gps::Gps::new(
            gpioe.pe8, gpioe.pe7, dp.UART7, prec.UART7, gpiog.pg1, gpiog.pg0, &clocks,
        );
```

ostenning, 2021-01-12:
Ah, thank you so much.

closed 2021-01-12 by ostenning