stm32-rs/stm32h7xx-hal #197: Ethernet DMA ring management: Gratuitous interrupt-free regions?

dnadlinger, 2021-03-04:
The Ethernet DMA driver places the memory barrier and descriptor tail buffer writes into an interrupt-free region, e.g.:

https://github.com/stm32-rs/stm32h7xx-hal/blob/4ca09fa94af0fed8084633c142767cf6515ed8de/src/ethernet/eth.rs#L322-L331

We didn't have this in the quartiq/stabilizer driver, and I can't think of a good reason for it – what's the story here?

richardeoin, 2021-03-07:
I can't think of a good rationale for these interrupt-free regions either. The write to `_dtpr` should correspond to a single [ptr::write_volatile](https://doc.rust-lang.org/std/ptr/fn.write_volatile.html) call, and I don't see how that could ever be split into multiple stores. 

There's also interrupt-free regions in the DMA channel initialisation ([here](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/ethernet/eth.rs#L127) and [here](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/ethernet/eth.rs#L277)), which I think are also unnecessary since the `s[tr]` bits in the `damc[tr]x_cr` registers aren't set until [later](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/ethernet/eth.rs#L632).

These could just have been an overly-defensive idea added during development that never got removed. If you're able to successfully test without them then removal is probably the right answer.

dnadlinger, 2021-03-07:
We are actually using this (or rather, one of the progenitor implementations in quartiq/stabilizer, before that switched to stm32h7xx-hal) without those interrupt-free regions in production. They must have been added later at some point, but I really can't see what the point would be – the DMA engine will continue to run even if interrupts are disabled… I'll post a PR to remove them.

commit c92baff8 (2021-03-07): Ethernet: Remove extraneous interrupt-free regions

There is no gain from disabling interrupts between the memory barriers
and writing to the DMA engine descriptor ring pointers - even if an
interrupt is executed between the barrier and the atomic store to the
pointer, it doesn't make the descriptors any less valid.

Interrupts are also irrelevant for transmit/receive ring initialisation,
which needs to run wiht the DMA engine disabled anyway.

GitHub: Fixes #197.
commit 5c054402 (2021-03-07): Merge #198

198: Ethernet: Remove extraneous interrupt-free regions r=richardeoin a=dnadlinger

There is no gain from disabling interrupts between the memory barriers
and writing to the DMA engine descriptor ring pointers - even if an
interrupt is executed between the barrier and the atomic store to the
pointer, it doesn't make the descriptors any less valid.

Interrupts are also irrelevant for transmit/receive ring initialisation,
which needs to run wiht the DMA engine disabled anyway.

GitHub: Fixes #197.
PR #198 (2021-03-07): Ethernet: Remove extraneous interrupt-free regions
There is no gain from disabling interrupts between the memory barriers
and writing to the DMA engine descriptor ring pointers - even if an
interrupt is executed between the barrier and the atomic store to the
pointer, it doesn't make the descriptors any less valid.

Interrupts are also irrelevant for transmit/receive ring initialisation,
which needs to run wiht the DMA engine disabled anyway.

GitHub: Fixes #197.
closed 2021-03-07 by bors[bot]