stm32-rs/stm32h7xx-hal #203: transmute between types of different sizes

pdgilbert, 2021-03-12:
Not sure what changed but this just started coming up in one of my CI tests
```

Compiling stm32h7xx-hal v0.8.0
error[E0512]: cannot transmute between types of different sizes, or dependently-sized types
   --> /home/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7xx-hal-0.8.0/src/rng.rs:110:67
    |
110 |                         let bytes: [$type; BATCH_SIZE] = unsafe { mem::transmute(random_word) };
    |                                                                   ^^^^^^^^^^^^^^
...
126 | rng_core!(usize);
    | ----------------- in this macro invocation
    |
    = note: source type: `u32` (32 bits)
    = note: target type: `[usize; 0]` (0 bits)
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)
```
I have been messing with my code, but pretty sure my changes didn't cause this.

mattico, 2021-03-13:
What's your rustc version?

mattico, 2021-03-13:
Found it: https://github.com/pdgilbert/rust-radio-sx127x/runs/2099019706 Rust 1.50.0

mattico, 2021-03-13:
This is caused by building stm32h7xx-hal for `x86_64-unknown-linux-gnu`. Be sure to use `--target=thumbv7em-none-eabihf`.

I've been confused by this before, we should add a compile error for this.

pdgilbert, 2021-03-13:
I can verify that adding  `--target=thumbv7em-none-eabihf` fixes the problem. But I was building with 

```
cargo build --no-default-features --release --features stm32h742
```
so not specifying any target for this step. That has worked previously and still works with other hals as you can see at https://github.com/pdgilbert/rust-radio-sx127x/actions/runs/647552905.
So it seems to be picking up a default target somewhere. Do I now have to start specifying the target for this first build step?

mattico, 2021-03-13:
It looks like [the last CI run that worked](https://github.com/pdgilbert/rust-radio-sx127x/runs/2084207777?check_suite_focus=true) didn't have this "build" step, just the "build example" steps which use cross and specify the target.

If you don't specify the target the default target is the target of the host, in this case `x86_64-unknown-linux-gnu`.

> Do I now have to start specifying the target for this first build step?

I'd say so, yes. Is there a reason that you're building the HAL for Linux? It wouldn't be hard to fix that build but I hadn't considered that doing so would be intentional and not a mistake.

pdgilbert, 2021-03-13:
(I was so blurry eyed from much workflow tweaking that I could not see what was happening.) I think the problem may be an overly agressive build strategy rather than the need for an error message. Here is a summary for anyone interested.

None of the following attempt to build the target directory, eg target/thumbv7em-none-eabihf

```
  cargo build  --no-default-features 
```

compiles some pieces but not anything requiring features.  (caveat. Above works in rust-radio-sx127x which is using embedded-hal 1.0.0-alpha.4. In another package using embedded-hal 0.2.4 I get the usual problem  `no InputPin  in  digital::v2` until I specify some features.) The follow all work in both packages  and compile more pieces:

```
  cargo build  --no-default-features  --features=stm32f0xx,stm32f030xc
  cargo build  --no-default-features  --features=stm32f4xx,stm32f411
  cargo build  --no-default-features  --features=stm32f1xx,stm32f103
  cargo build  --no-default-features  --features=stm32f3xx,stm32f303xc
  cargo build  --no-default-features  --features=stm32f4xx,stm32f411 
  cargo build  --no-default-features  --features=stm32f7xx,stm32f722
  cargo build  --no-default-features  --features=stm32l0xx,stm32l0x2
  cargo build  --no-default-features  --features=stm32l1xx,stm32l100
  cargo build  --no-default-features  --features=stm32l4xx,stm32l4x2
```
  
But then 

```
  cargo build  --no-default-features  --features=stm32h7xx,stm32h742
```

fails with

```
error: could not compile `stm32h7`

Caused by:
  process didn't exit successfully: `rustc --crate-name stm32h7 --edition=2018 /home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7-0.12.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="cortex-m-rt"' --cfg 'feature="default"' --cfg 'feature="rt"' --cfg 'feature="stm32h743"' -C metadata=92009c8beecd4dbb -C extra-filename=-92009c8beecd4dbb --out-dir /home/paul/githubClones/rust-radio-sx127x/target/debug/deps -L dependency=/home/paul/githubClones/rust-radio-sx127x/target/debug/deps --extern bare_metal=/home/paul/githubClones/rust-radio-sx127x/target/debug/deps/libbare_metal-efa5ff3a789aa0d1.rmeta --extern cortex_m=/home/paul/githubClones/rust-radio-sx127x/target/debug/deps/libcortex_m-5c544b4361cb95da.rmeta --extern cortex_m_rt=/home/paul/githubClones/rust-radio-sx127x/target/debug/deps/libcortex_m_rt-571e73a6ac779bfe.rmeta --extern vcell=/home/paul/githubClones/rust-radio-sx127x/target/debug/deps/libvcell-00c45049692fe545.rmeta --cap-lints allow -L /home/paul/githubClones/rust-radio-sx127x/target/debug/build/stm32h7-ba81181585b94d3c/out -L /home/paul/githubClones/rust-radio-sx127x/target/debug/build/cortex-m-rt-1ec2c44a7897cd63/out` (signal: 9, SIGKILL: kill)
 ```
 
So is it trying to build the target for the default triple, or doing something else that requires knowing the target? 

(I'm new to embedded but the default does not make sense to me. Is there ever any situation one would build an embedded no_std hal like stm32h7xx for the compiler's host? I certainly was not doing it on purpose.)

BTW, specifying the target is good advice if you just want things to work.

mattico, 2021-03-13:
> So is it trying to build the target for the default triple, or doing something else that requires knowing the target?

Yes, it is building for the host target. Unless you tell cargo to use a different target via command line, environment variable, or `.cargo/config` it will always build for the host target. There is no automatic detection, there is no way for libraries to tell cargo what target to use. Some libraries will set the target in their `.cargo/config` file but that only affects cargo invocations in their directory, e.g. examples.

I think the root of the confusion is that the HALs would build for x86 at all, aren't they hardware specific? Yes, they are, and they won't *run* on x86. However, most of the hardware specific bits in the HALs are just reading and writing to specific memory locations which won't *work* on other hardware but it will build just fine. Unless the library uses assembly instructions or relies on e.g. pointer size it's easy to accidentally make something which can build on x86. There are also reasons to purposely make it work, like [being able to run non-target-specific tests](https://github.com/stm32-rs/stm32h7xx-hal/blob/ad2a1ae9831dd34e171cc178cdad84e770b4c0ae/src/i2c.rs#L757) on your host machine.

This specific error is caused by [a calculation that assumes the pointer size is 32-bits](https://github.com/stm32-rs/stm32h7xx-hal/blob/451251accd66bd7d9965b511cf2c97d62bfc4806/src/rng.rs#L94-L126) which is probably not too common among HALs, at least not in a way which causes a compile-time error.

pdgilbert, 2021-03-14:
Ok, thank you for the explanations. I think I mostly understand this. But saying "the default target is the target of the host" suggests to me that on my computer

```
cargo build  --no-default-features  
```
and

```
cargo build  --no-default-features  --target x86_64-unknown-linux-gnu
```
should do the same thing. But the second creates a directory `target/x86_64-unknown-linux-gnu` and puts stuff in it, so you must mean something slightly different by "default target".

mattico, 2021-03-14:
That appears to be a bug/simplification in cargo. [Judging by the code and comments](https://github.com/rust-lang/cargo/blob/5c74eb5707deb2c87d3bcad21e61af2e21f0b555/src/cargo/core/compiler/compile_kind.rs#L12-L44) when cargo is given a `--target` argument it is considered cross compiling and it simply passes along the target to rustc and builds in a subdirectory that contains the target name.

You may think that this logic would fix that:

```rust
    pub fn for_target(self, target: &Target) -> CompileKind {
        // Once we start compiling for the `Host` kind we continue doing so, but
        // if we are a `Target` kind and then we start compiling for a target
        // that needs to be on the host we lift ourselves up to `Host`.
        match self {
            CompileKind::Host => CompileKind::Host,
            CompileKind::Target(_) if target.for_host() => CompileKind::Host,
            CompileKind::Target(n) => CompileKind::Target(n),
        }
    }
```

but no, [for_host is only true for things that need to be built for the host](https://github.com/rust-lang/cargo/blob/adb35807a3643962f27b428a05485d9cc8fc0ae0/src/cargo/core/manifest.rs#L761-L764) like [plugins, proc-macros and build scripts](https://github.com/rust-lang/cargo/blob/414ccb5d491030268e4c7b87dc2345d73760ecdf/src/cargo/util/toml/targets.rs#L782-L786). From my cursory glance it appears that cargo doesn't even know what the host target is, I can't see it stored anywhere obvious.

pdgilbert, 2021-03-14:
Thanks, that explains it.

commit 443eacdc (2021-03-16): Merge #204

204: Add compile error reminding to build with the correct target r=richardeoin a=mattico

Fixes #203
PR #204 (2021-03-16): Add compile error reminding to build with the correct target
Fixes #203
closed 2021-03-16 by bors[bot]