stm32-rs/stm32h7xx-hal #207: ethernet latency - what should we expect?

astraw, 2021-03-26:
I have just incorporated ethernet into my RTIC-based project on the h7 Nucleo board and am wondering what kind of latency performance I should expect with ethernet. I haven't dug deep into the code or the reference manual yet. If I just "ping" my nucleo I get a round trip time of < 1msec. I presume this is hitting the smoltcp stack on the chip but perhaps it is handled purely at the hardware level? If I have a higher level ping command which deserializes some input, reserializes it and sends a packet back, I can get latencies of as low as 9 msec but frequently also spiking up to almost 2 seconds. This also happens when I have disabled most other parts of the firmware.

Are these numbers similar to what others might expect? Is the high jitter (from 9 msec to almost 2000 msec) something that may reflect a fundamental aspect of the hardware, perhaps a suboptimal implementation in stm32h7xx-hal, or problems with my own code?

adamgreig, 2021-03-27:
If the stm32 is on a wired LAN with your computer, I'd expect all latencies to be well below 1ms; even on wifi you shouldn't see more than a few ms. The ICMP ping processing is entirely handled by smoltcp, not the hardware.

A quick test using an stm32h745 nucleo board and the ethernet example from the HAL gets ICMP pings of avg 0.09ms/max 0.14ms, and if I add a UDP socket to the example that just replies with whatever you send it, I see avg 0.13/max 0.16ms.

Maybe you're not calling `poll()` often enough? My guess is that it's only being called on ethernet reception interrupt, so you poll after receiving a packet, enqueue a response for transmission, but never call poll again to transmit it - until another packet comes in (e.g. random network broadcast), triggering the outgoing packet to be sent. Do you find your latency gets much better if you leave a ping running at the same time, e.g. with a very small ping interval (sudo ping -f, or ping -i 0.2 or something)?

astraw, 2021-03-27:
Thank you! I just tested and, yes, I was simply not polling enough. Simply adding a poll after calling `socket.send_slice()` makes a big difference. I am now consistently getting round trip times to be very consistently between 6 and 8 msec and I haven't started to optimize any of the other code. (This is with wired LAN, so according to your numbers there is still quite some room for improvement.)

closed 2021-03-27 by astraw