stm32-rs/stm32h7xx-hal #212: Add more description for examples

ostenning, 2021-04-12:
I think it would be useful for people attempting to get various examples working to have description of the board that the example was originally written for at the top of the example.

In my case, I'm attempting to get SDMMC working and so I'm copying the example but it doesn't work for me. Knowing what board the example was originally written for can give clarity for people.

@richardeoin which board was the sdmmc example written for?

Thanks

richardeoin, 2021-04-13:
I'd definitely support anything to make the examples more accessible! And adding board descriptions would be a nice improvement, although I think some examples can still be valuable without necessarily applying to a specific dev board.
For the sdmmc example, I think it was written for a STM32H747I-DISCO

robamu, 2021-11-01:
I think its a general problem that these examples are board specific in many cases. Also, its tricky to get started without any of the additional helper files like `openocd` files which make the workflow similar to what is taught in the embedded rust book. Making the examples board specific helps a lot, but maybe it would also be useful so specify how to use the examples the way it was done [here](https://github.com/stm32-rs/stm32f1xx-hal) or [here](https://github.com/stm32-rs/stm32f4xx-hal) . This includes writing down all depencies. I just copied the `utilities` folder and the blinky.rs script into my example project which is set up for convenient `cargo run` flashing but I still get dependencies error