stm32-rs/stm32h7xx-hal #217: Build failes with features = [ "stm32h743", "smps" ]

daniel-dbg-ginsburg, 2021-04-23:
When configured for [ "stm32h743", "smps" ], compilation fails with:
error[E0599]: no method named `sden` found for struct `stm32h7::R<u32, Reg<u32, stm32h7::stm32h743::pwr::_CR3>>` in the current scope
   --> /home/dbg/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7xx-hal-0.9.0/src/pwr.rs:176:44
    |
176 |                 assert!(self.rb.cr3.read().sden().bit_is_clear(), error);
    |                                            ^^^^ method not found in `stm32h7::R<u32, Reg<u32, stm32h7::stm32h743::pwr::_CR3>>`

Versions: 
stm32h7xx-hal v0.9.0
stm32h7 v0.13.0

The reason for the failure is that the PWR_CR3 in stm32h743 does not have the SDEN bit.

mattico, 2021-04-23:
I can't find any STM32H743 models that include an SMPS. Which chip are you using?

daniel-dbg-ginsburg, 2021-04-23:
I apologize, I didn't make my point clear. Compilation should fail with a more meaningful message than a bunch of errors about a non-existing flag in a configuration register.

richardeoin, 2021-04-25:
It is indeed impossible to hit this with any H7 product from ST, but fixed anyhow for completeness

commit d511e017 (2021-05-10): Add explict error message when both rm0433 and smps are selected

Fixes #217
commit a0e89166 (2021-05-10): Merge #218 #223

218: Add explict error message when both rm0433 and smps are selected r=richardeoin a=richardeoin

Fixes #217

223: Satisfy clippy by initialising LateResources in (alphabetical?) order r=richardeoin a=richardeoin

https://rust-lang.github.io/rust-clippy/master/index.html#inconsistent_struct_constructor
PR #218 (2021-05-10): Add explict error message when both rm0433 and smps are selected
Fixes #217
closed 2021-05-10 by bors[bot]