stm32-rs/stm32h7xx-hal #228: Initializing a DMA peripheral breaks the other DMA's streams

mattico, 2021-06-11:
1. Initialize DMA2 (`StreamsTuple::new()`).
2. Configure DMA2 Stream2 to transmit over a UART, and start it.
3. Initialize DMA1.
4. DMA2 Stream2 is now broken. If it was in the middle of transmitting it stops and never completes. If you try to start transmitting again it never actually sends any bytes to the UART.

I think this is a silicon bug. I looked through every relevant piece of code during the long debugging session, and checked all the error flags for everything.

Either way we should prevent this from happening.

mattico, 2021-06-11:
See the repro here: https://github.com/mattico/stm32h7xx-hal/commit/61807dd5b2986b9cdbd01b153bed00dd1378818e

The issue also occurs with the SPI peripheral so it's not UART specific.

richardeoin, 2021-06-13:
Thanks for reporting this one. It looks like a bit of a pain, but probably related to the reset itself.

I can reproduce it (actually on a STM32H735 since I had one set up). Removing the [write 1 on RCC AHB1RSTR DMA2RST ](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/dma/dma.rs#L298) solves the problem, and incidentally writing a 1 to AHB4RSTR BDMARST doesn't affect anything.

Note that writing 1 to DMA2RST doesn't actually reset DMA1 like writing 1 to DMA1RST would, it just kills the transfers. In fact the registers of the 'victim' DMA appear to be unaffected. I'm therefore inclined to agree that there's some unintended cross-coupling between the resets of DMA1 and DMA2 that isn't mentioned in any ST documentation. If that's true, then indeed it's a case of preventing this using the type system and explaining why in the docs.

A good workaround for most situations would be to use multiple streams on the same DMA. Stabilizer has a great [example of this](https://github.com/quartiq/stabilizer/blob/master/src/hardware/setup.rs) that uses all 8 streams on DMA1. Another workaround is to call `StreamsTuple::new` on both DMA1 and DMA2 before creating any transfers - that also works fine.

mattico, 2021-06-13:
One workaround is to enable both DMAs before using them, which is what I'm doing. We could enforce that in the API, though some may not want both enabled (power?).

We could also just not reset the DMAs. They should already be reset by power-on unless a user deliberately wrote some DMA registers, no? I say let them do it.

richardeoin, 2021-06-15:
That's correct, the registers are certainly in their default values after a power-on reset. In general I would like the use of peripheral resets / enables to be consistent within the HAL, but indeed removing the peripheral DMA reset is an easy and effective solution here.

commit c84d9875 (2021-06-26): Don't reset the peripheral in the DMA1/2 StreamsTuple::new method

Closes #228
commit 95667cd7 (2021-06-26): Merge #229

229: Don't reset the peripheral in the DMA1/2 StreamsTuple::new method r=richardeoin a=richardeoin

Closes #228
PR #229 (2021-06-26): Don't reset the peripheral in the DMA1/2 StreamsTuple::new method
Closes #228
closed 2021-06-26 by bors[bot]