stm32-rs/stm32h7xx-hal #238: i2c timing calculation overflows SCLDEL

davidlattimore, 2021-08-18:
If my peripheral clock is running at 80MHz and I configure i2c to run at 1MHz, the code for computing i2c timing produces the following values:

```
presc_reg = 0
scll = 51
sclh = 25
sdadel = 10
scldel = 19
```

But SCLDEL and SDADEL are only 4 bits each, so can't have values over 15.

I think the problem stems from the selection of `presc_reg`:

```rust
let presc_reg = ((ratio - 1) / 384) as u8;
```

Dividing by 384 gives very precise control over the frequency of SCL, but severely limits the range of SCLDEL and SDADEL. If we change this to divide by a smaller number, then tests fail because the frequency is less precise than the tests demand.

I'm actually using an stm32l4, which has more or less identical code for computing i2c timing. It seems like there are lots of copies of the code in various different hal crates. But this crate has tests!

Do you think it might make sense to move this code into a common crate that can be shared between the various hal crates that have similar (or identical?) i2c timing registers?

Also, do you have any thoughts on how to fix the problem? I'm wondering if i2c frequency needs to be so accurate, or if we could just pick the highest frequency <= to the requested frequency that can be achieved with the selected prescaler. Another option would be to loop over all 16 possible prescalers to see which one produced the best result - then hope that the code gets inlined into a function that knows the peripheral clock speed and can be optimized away.

richardeoin, 2021-08-18:
Hi, Thanks for reporting! This is an issue for sure, and it's possible for this overflow to occur on SDADEL also. As a first step toward solving it, the i2c timing calculation should clamp the values to 15 (like it [clamps the minimum value](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/i2c.rs#L260-L261)) and the tests should test this!

I see how the small prescaler values often leads to SCLDEL values greater than 15. I don't see an issue with reducing that scale factor to 96 for example, and tolerate a bit more error (2% high in frequency would be acceptable imo). Btw there's no real calculation behind the constants [here](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/i2c.rs#L209-L210), they're just round values that pass the tests.

I would support a common crate for this code (I've not actually tracked where all the copies are..), but there's a bit of overhead to support the crate and submit PRs to all the various HALs. I think that's why copy+paste has been the favoured approach so far.

richardeoin, 2021-08-18:
I made a first attempt at fixing this on a branch [here](https://github.com/richardeoin/stm32h7xx-hal/commit/d8e4ea66c2514f3d23126ff0f2ee58d280dd8f3a). There may be a better solution however.

davidlattimore, 2021-08-19:
Great! I tried your new code (on an stm32l452). I tried running the i2c speed at speeds between 10KHz and 1MHz. It seemed to work except for some lower speeds between 11 and 27Khz, but it's quite likely that's due to the device I'm talking to, not due to the timing configuration used. Indeed, I tried using a timing configuration produced by STM32CubeMX and it failed at these speeds as well.

closed 2021-09-14 by bors[bot]

David-OConnor, 2021-09-16:
I solved using a different approach here, based on the RM tables: https://github.com/David-OConnor/stm32-hal/blob/main/src/i2c.rs#L275

commit 1b1205e6 (2021-09-14): Clamp scldel/sdadel to 4-bit and update divider calculations

Previously this code was calculating scldel/sdadel values > 15 which were
correct, but could not be successfully used by the hardware.

Update divider calculations to generally pick a larger prescaler value,
resulting in a slower scaled clock. This means that scldel/sdadel values up to
15 still result in the required delays

Fixes #238
commit fb66f6b5 (2021-09-14): Merge #245 #252

245: Implement DMA TargetAddress for I2C[123] r=richardeoin a=richardeoin

252: Clamp scldel/sdadel to 4-bit and update divider calculations r=richardeoin a=richardeoin

Previously this code was calculating scldel/sdadel values > 15 which were correct, but could not be successfully used by the hardware.

Update divider calculations to generally pick a larger prescaler value, resulting in a slower scaled clock. This means that scldel/sdadel values up to 15 still result in the required delays.

Fixes #238
PR #252 (2021-09-14): Clamp scldel/sdadel to 4-bit and update divider calculations
Previously this code was calculating scldel/sdadel values > 15 which were correct, but could not be successfully used by the hardware.

Update divider calculations to generally pick a larger prescaler value, resulting in a slower scaled clock. This means that scldel/sdadel values up to 15 still result in the required delays.

Fixes #238