stm32-rs/stm32h7xx-hal #243: [feature request] API for configuring LTDC "line interrupt" / VSYNC

starthal, 2021-09-02:
The "Line" interrupt (`LIF`/`LIE`) can be used as a VSYNC signal.

RM0455:
> Line interrupt: generated when a programmed line is reached. The line interrupt
position is programmed in the LTDC_LIPCR register

In ST TouchGFX code, a VSYNC interrupt is configured as such:
 - Read the back porch line from `LTDC_BPCR` (previously configured)
 - Write `back_porch_line - 1` to `LTDC_LIPCR.LIPOS`
 - Enable the interrupt: set `LTDC_IER.LIE`

The API could provide `LIPOS` and `LIE` configuration separately, and/or provide a convenience function for this VSYNC-like interrupt.

NOTE: The RM does not seem to clarify whether the interrupt happens at the beginning or the end of the line scan. Presumably the off-by-one configuration in the ST code gives the correct behavior.

starthal, 2021-09-02:
My team could do a PR for this; we'd like advice for the appropriate API.

richardeoin, 2021-09-02:
It would be great to support this! There is very rudimentary support for one interrupt in the LTDC HAL already, but it's in need of improvement and breaking changes are welcome in order to improve the API.  The [SPI HAL has a good example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/spi.rs#L805-L855) of a more mature set of functions for managing interrupts. 

I'd say either approach to managing the `LIPCOS` value is acceptable, but my suggestion would be to set `LIPOS` to `back_porch_line - 1` within the `init` method. That would be equally convenient for the user, but avoids any 'surprising' behaviour in the interrupt methods.

You could also create a separate method that writes to `LIPCR.LIPOS`, if there is a use case for setting any value other than `back_porch_line - 1`.