stm32-rs/stm32h7xx-hal #270: DCMI HAL Support Feature

folknology, 2021-10-31:
Are there any plans for the HAL to support DCMI for the camera interface? I notice it is currently absent is that for an specific reason like lack of documentation or SVD support or just no one has gotten around to it yet?

richardeoin, 2021-10-31:
Hi! I don't know of any development related to DCMI support so far, but contributions are very welcome! It's mostly that no one has gotten around to it, since the SVD support looks OK. 

It's worth a quick look at the crates for families (F7, G4 etc..) that might share the peripheral in case anyone has started it there, but in this case I think you're out of luck :(

folknology, 2021-10-31:
Thanks @richardeoin I have a H750 included in a board design I need to work on next year and I am leaving the DCMI pins available to investigate, I may well return to it then (when I have more embedded rust under my belt with current F7 design).