stm32-rs/stm32h7xx-hal #275: Ethernet interrupts generated spuriously after heavy traffic

ryan-summers, 2021-11-09:
Refer to https://github.com/quartiq/stabilizer/issues/450 for further information, but it appears that the ETH_MAC has a number of interrupts enabled by-default.

Specifically, interrupts related to MMC counters will occur, for example, after 2^31 ethernet packets are transmitted. If the ethernet ISR is enabled, this can result in applications hanging if they use the default `ethernet_handler()`, as it does not robustly clear all ISR flags.

ryan-summers, 2021-11-09:
I believe we need to mask away MMC RX/TX interrupts in the `ETH_MMC_TX_INTERRUPT_MASK` and `ETH_MMC_RX_INTERRUPT_MASK` registers. These interrupts are unmasked by default.

jordens, 2021-12-13:
@ryan-summers @richardeoin can we squeeze [the fix for this](https://github.com/quartiq/stabilizer/pull/462/files) into v0.11.0?

richardeoin, 2021-12-13:
@jordens Agreed! PR above

commit cddf0e0c (2021-12-13): Disable ethernet MAC MMC counter interrupts

Patch from https://github.com/quartiq/stabilizer/pull/462

Tested that the expected registers are indeed set on a STM32H747I-DISCO

> MMC_RX_INTERRUPT_MASK:               0x0C020060  MMC Rx interrupt mask register
> MMC_TX_INTERRUPT_MASK:               0x0C20C000  MMC Tx interrupt mask register

Closes #275
commit 6ce89dff (2021-12-13): Merge #295

295: Disable ethernet MAC MMC counter interrupts r=jordens a=richardeoin

Patch from https://github.com/quartiq/stabilizer/pull/462

Tested that the expected registers are indeed set on a STM32H747I-DISCO

> MMC_RX_INTERRUPT_MASK:               0x0C020060  MMC Rx interrupt mask register
> MMC_TX_INTERRUPT_MASK:               0x0C20C000  MMC Tx interrupt mask register

Closes #275
PR #295 (2021-12-13): Disable ethernet MAC MMC counter interrupts
Patch from https://github.com/quartiq/stabilizer/pull/462

Tested that the expected registers are indeed set on a STM32H747I-DISCO

> MMC_RX_INTERRUPT_MASK:               0x0C020060  MMC Rx interrupt mask register
> MMC_TX_INTERRUPT_MASK:               0x0C20C000  MMC Tx interrupt mask register

Closes #275
closed 2021-12-13 by bors[bot]