stm32-rs/stm32h7xx-hal #285: OTG_HS interrupt possibly missing from stm32h7xx_hal::interrupt enum for STM32H723ZG

TomDeRybel, 2021-12-02:
I'm porting over some code to a ST Nucleo-H723ZG board and ran into a little snag, shown below. The other interrupts used in the code compile fine (eg: EXTI15_10, TIM1_UP).

The OTG_HS interrupt is listed on the documentation website and worked fine on the H743 I used before.
https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/enum.interrupt.html#variant.OTG_HS

`error[E0599]: no variant or associated item named `OTG_HS` found for enum `stm32h7xx_hal::interrupt` in the current scope
   --> src/main.rs:251:17
    |
251 |     #[task(binds = OTG_HS, shared = [usb_dev, usb_cdc])]
    |                    ^^^^^^ variant or associated item not found in `stm32h7xx_hal::interrupt`
`

I'm using the following versions/cargo lines:
`stm32h7 = { git = "https://github.com/stm32-rs/stm32-rs-nightlies", features = ["stm32h735","rt"] }
stm32h7xx-hal = { git = "https://github.com/stm32-rs/stm32h7xx-hal", features = ["stm32h735","rt","usb_hs"] }`

As an aside, some register field names seem to have changed between the H743 and H725. Not sure if this is intentional:
H743
`ctx.device.DBGMCU.cr.modify(|_, w| {
			w.dbgsleep_d1().set_bit();
			w.dbgstby_d1().set_bit();
			w.dbgstop_d1().set_bit()
		});`

H723
`ctx.device.DBGMCU.cr.modify(|_, w| {
			w.dbgslpd1().set_bit();
			w.dbgstbd1().set_bit();
			w.dbgstpd1().set_bit()
		});`

Thank you for all the work done on these chips! These crates make them so much more friendly to use.

mattico, 2021-12-02:
The documentation on docs.rs is generated for the stm32h743v MCU. You can run cargo doc locally to generate documentation for the MCU you are using.

Both of those issues are in https://github.com/stm32-rs/stm32-rs, and should be relatively straightforward to fix.

TomDeRybel, 2021-12-03:
Thank you for the fast reply! 

Indeed, after running cargo doc, those interrupts are missing from the generated documentation. I have submitted the issue with the stm32-rs project.

TomDeRybel, 2022-07-05:
The issue should be officially fixed now in stm32-rs: https://github.com/stm32-rs/stm32-rs/pull/677
and is included in release 0.15.1
So, I assume that everything will follow when stm32h7xx-hal updates to that never version in due course.

richardeoin, 2022-08-03:
Great to hear, thanks for sharing @TomDeRybel

closed 2022-08-03 by richardeoin