stm32-rs/stm32h7xx-hal #315: [Feature Request] Fully type erased pin

jg2562, 2022-02-11:
Currently, there is only the ability to partially erase the pin's typing, meaning pins still must remain within the port they originated from. However, this restricts the usage if there requires an array that crosses port boundaries. I propose implementing a pin with the origin port type fully erased similar to how [stm32f4xx-hal does it](https://github.com/stm32-rs/stm32f4xx-hal/blob/8ab1f8fed69d8cded57f0b3a3b12a1e9c7d479be/src/gpio/erased.rs). How feasible would this be?

mlamoore, 2022-02-12:
Looking at the reference manual for the STM32H742/743/750/753, the GPIO
registers are arranged with uniform layouts and offsets so the same
technique would be possible there.

If you wanted to do this, I would recommend starting by looking through all
the supported reference manuals and make sure they all use consecutive GPIO
register blocks 0x400 apart in memory like the 742/743/750/753, then port
this over.

jg2562, 2022-02-13:
Yeh, I will take a look at that, thank you for the direction!