stm32-rs/stm32h7xx-hal #323: [Feature Request] Reconfigurable serial ports

jg2562, 2022-02-25:
I wanted to be able to reconfigure the baud rate on a particular serial port after creation, but I didn't find a way to do it safely. It seems like it may be useful to be able to set a new config after creation, however, I'm not sure how feasible this is.

mlamoore, 2022-02-25:
That sounds feasible to me; you would need to write a new method on
Serial<UART> that consumes Self, I think, and make sure to do all the
required register writes according to the reference manual to stop the UART
then start it with the required settings.

You might need to also take the clock config or other parameters from the
original UART initialization too...

Alternately, you could just write a new method that destroys a UART,
returning the underlying peripheral and pins, and let the user create it
fresh with the existing initialization. That would really require all the
initialization, though, even the RCC handle...

However, I think the HAL is not well suited to changing the main clock
configuration (core clocks, PLLs, etc) after initialization; we assume that
you freeze the clocks then initialize other peripherals with those clocks
that don't change. Changing UART baud rate should be straightforward,
changing the PLL/clock source that feeds the UART would not be.

olback, 2022-02-25:
@mlamoore

> a new method on Serial<UART> that consumes Self

Why wouldn't `&mut self` be enough in this case? I feel like it would be much easier to use.

mlamoore, 2022-02-25:
> a new method on Serial that consumes Self

I don't have any concrete reason why &mut self wouldn't work, especially
for just changing the baud rate or similar properties.

I was thinking consuming self seemed more similar to how the GPIO and PWM
types change configuration, but those are using type state programming with
the configuration being part of the type so they need to return a different
type.

In this case, &mut self probably makes more sense.

jg2562, 2022-02-25:
Oh awesome that was exactly what i was looking for! Thanks!

As a side note, what would be the trade-off to adding a method like `serial.release()` that gave back the pins and the peripheral?

mattico, 2022-02-25:
In case you don't know, there is a `serial.release()` already but it doesn't return the pins. The serial port structs don't contain the pins, so we'd have to add the pin types to the generic type parameters, which would be very annoying in some situations.

jg2562, 2022-02-27:
Yeh i did, but i wasnt sure how the type parameter encoding would work, but that explains it, so thank you!

mattico, 2022-02-28:
I should mention: the workaround is to use [`serial_unchecked`](https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/serial/trait.SerialExt.html#tymethod.serial_unchecked) which doesn't require you to input the pins as arguments.

commit 4fe01a8c (2022-02-28): Merge #324

324: serial: support reconfiguring ports, Tx/Rx pin swap, and join() for Tx/Rx r=richardeoin a=mattico

Fixes #323
PR #324 (2022-02-28): serial: support reconfiguring ports, Tx/Rx pin swap, and join() for Tx/Rx
Fixes #323