stm32-rs/stm32h7xx-hal #329: [Feature Request] DMA configuration for SAI peripherals does not include all possible cases

backtail, 2022-03-01:
I recently came across (in my specific case) a bug. Here, you'll find the details: https://github.com/antoinevg/daisy_bsp/issues/8#issuecomment-1054506275

What it boils down to is that the DMA config for all SAI peripherals are "hardwired" in the hal library.

https://github.com/stm32-rs/stm32h7xx-hal/blob/72adb08daf594d623511d652647c23088b54ef12/src/dma/dma.rs#L938-L941

Channel A is always configured as TX and channel B is always configured as RX even though the individual I²S channels can be configured either way. 

https://github.com/stm32-rs/stm32h7xx-hal/blob/72adb08daf594d623511d652647c23088b54ef12/src/sai/i2s.rs#L73-L76
https://github.com/stm32-rs/stm32h7xx-hal/blob/72adb08daf594d623511d652647c23088b54ef12/src/sai/i2s.rs#L177-L178

This means if one would need to use channel A of the SAI as RX channel and vice versa for channel B, it's currently not possible with the DMA.

richardeoin, 2022-03-01:
Thanks for reporting! The DMA support in this HAL is still relatively immature, and I think the Rust community is still figuring out how the best approach to DMA. So problems like this one are good to highlight where improvements (or new approaches) are needed.

SAI breaks the assumption that each peripheral has a single address for DMA write and a single address for DMA read. As you've noticed, the current implementation re-imposes that assumption that by fixing one channel as read and one channel as write.

Right now the init call does not specify either Channel A or Channel B. Currently we use type inference to select a single [`TargetAddress` ](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/dma/traits.rs#L393) implementation based on just `SAI1` and `dma::MemoryToPeripheral`, and therefore that implementation is hard-coded as Channel A.
```rust
let dma1_str0: dma::Transfer<_, _, dma::MemoryToPeripheral, _, _> = dma::Transfer::init(
    dma1_streams.0,
    unsafe { pac::Peripherals::steal().SAI1 },
    tx_buffer,
    None,
    dma_config,
);
```

My first idea for fixing this is to implement `TargetAddress` for a compound type that represents both the SAI peripheral and the channel. The implementation would look like this.

```rust
peripheral_target_address!(
    ((pac::SAI1, ChannelANewMarkerType), cha.dr, u32, M2P, DMAReq::SAI1A_DMA),
    ((pac::SAI1, ChannelANewMarkerType), cha.dr, u32, P2M, DMAReq::SAI1A_DMA),
    ...
);
```

This seems promising enough to be worth prototyping

richardeoin, 2022-03-01:
I've made an attempt at this - see the changes to the [sai_dma_passthrough.rs example](https://github.com/stm32-rs/stm32h7xx-hal/pull/330/files) on this branch #330

backtail, 2022-03-01:
Nice, this should work, hopefully. I will be able to test these changes on the weekend!

backtail, 2022-03-07:
Were you able to compile your changes?

David-OConnor, 2022-03-10:
Here's how I'd handle that:
```rust
let mut sai_tx = Sai::new(
        dp.SAI2,
        SaiConfig { // Block A
            mode: SaiMode::MasterReceiver,
            ..SaiConfig::i2s_preset()
        },
        SaiConfig { // Block B
            mode: SaiMode::MasterTransmitter,
            ..SaiConfig::i2s_preset()
        },
        &clock_cfg,
);
// ..
dma::mux(DmaChannel::C0, dma::DmaInput::Sai1A, &mut dp.DMAMUX1);
dma::mux(DmaChannel::C1, dma::DmaInput::Sai1B, &mut dp.DMAMUX1);

// ..

sai.write_dma(
    buf,
    SaiChannel::B,
    DmaChannel::C1,
    dma::ChannelCfg {
        circular: dma::Circular::Enabled,
        ..Default::default()
    },
    dma,
);

sai.read_dma(
    buf,
    SaiChannel::A,
    DmaChannel::C0,
    dma::ChannelCfg {
        circular: dma::Circular::Enabled,
        ..Default::default()
    },
    dma,
);

```

backtail, 2022-03-11:
I fixed my original problem (for that I created this issue) by bypassing HAL completely and just writing directly to the registers. But obviously it's not the most elegant solution. @richardeoi do you maybe know when your fix is going to be merged to the main branch? Until then my code is gonna look ugly :)

richardeoin, 2022-03-18:
@backtail Cargo definitely supports using a git source with the same version as the published crate (v0.11.0), so you shouldn't be having problems with that. In any case I'll finish the PR and merge it to master at some point.

commit 059b9579 (2022-03-27): Add ChannelA/B types for SAI, add DMA implementation

Allows implementing DMA P2M and M2P on both Channel A and Channel B. Closes #329
closed 2022-03-27 by bors[bot]