stm32-rs/stm32h7xx-hal #335: stm32h723zg "board crate not available"

imrank03, 2022-03-13:
I am working on the `stm32h723zg` board but the crate is not available for this board, can you please guide me how to create the crate for this board?

jg2562, 2022-03-15:
While this isn't a solution, I have been using the `stm32h735` feature instead and it has worked for the `stm32h723zg` so far.

imrank03, 2022-03-16:
> While this isn't a solution, I have been using the `stm32h735` feature instead and it has worked for the `stm32h723zg` so far.

@jg2562 Yes I am also using the stm32h735 feature, if I use flash write or erase getting panic or getting a hardfault. Any suggestions on how to use a flash driver.

hargoniX, 2022-03-16:
You have to obtain the svd files, merge them into stm32-rs/stm32-rs (probably with some patches), switch the PAC release to fetching directly from stm32-rs/stm32-rs until they do a release and then you should be able to integrate it into the feature list just how all the other PAC variants are.

imrank03, 2022-03-16:
@hargoniX Can you please guide me how to do it?

hargoniX, 2022-03-16:
Adding a new device is described in detail in the stm32-rs/stm32-rs README you can just follow that guide, as to where to obtain the SVD files for your specific chip my search engine is as good as yours really.

richardeoin, 2022-04-19:
Using the `stm32h735` feature for the STM32H723ZG should be a good solution for most cases.

If not, then `stm32h723` needs a separate target in the PAC first as discussed above

closed 2022-04-19 by richardeoin