stm32-rs/stm32h7xx-hal #338: Flash driver for stm32h723zg

imrank03, 2022-03-22:
Hello all, I am using a flash driver for `stm32h723zg` which is a `256 bit` data and address align, if I am trying to write a single byte it is not working. Can you help me with how to write single byte data in Flash? 

- My board architecture is `32 bit` but it is working on `32 byte` align data How is it?

ostenning, 2022-04-20:
Hi, the requirements for reading and writing FLASH are different. **Reading** requires that the buffer size and memory alignment is `32bit` and **writing** data requires a buffer size and memory alignment of `256bit`.
These are the limitations of writing/reading FLASH on the device. It is not possible to simply write 1 byte.
You can find further information from STM directly in the 0433 [Reference Manuals](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf).

My advice is to first read the data chunk you want, e.g. Read 256bits, then modify the single byte you want, and then write back the entire 256bits back to FLASH. The current HAL implementation wont do this for you.

@richardeoin do you think its necessary to support this kind of functionality?

Kind regards

imrank03, 2022-06-01:
Hi @ostenning,

I'm doing the same as you said, if I am trying to rewrite the bytes I have already written bytes, the controller is panicking, is there any way to write which is already written bytes in 32 byte address?

imrank03, 2022-07-01:
@ostenning: Thanks for your advice, it's working now. So I am closing this issue.

closed 2022-07-01 by imrank03