stm32-rs/stm32h7xx-hal #343: ADC example scaling wrong

jordens, 2022-03-28:
The ADC examples scale the ADC codes with `VREF/max_sample`.

https://github.com/stm32-rs/stm32h7xx-hal/blob/a46841c4465978aeddef3795b4f571c6c25cfcc9/examples/adc.rs#L76-L80

`max_sample = ((1 << bits) - 1) << lshift` by itself is correct according to its docstring. 

https://github.com/stm32-rs/stm32h7xx-hal/blob/a46841c4465978aeddef3795b4f571c6c25cfcc9/src/adc.rs#L790-L793

But the conversion is not. VREF corresponds to `1 << bits + lshift` here as in all ADCs that I know of.
c.f. https://github.com/stm32-rs/stm32f4xx-hal/issues/362#issuecomment-941014100 and other places.

The best approach is probably to change `max_sample()` and its docstring.

richardeoin, 2022-03-28:
I enjoyed the explaination on the f4 thread, thanks! Indeed this is clearly an error in the example, and a footgun in the API.

My preferred way of fixing this would be to add a new method, perhaps `pub fn slope(&self) -> u32` that returns `1 << bits + lshift`. After that, `max_sample` can be marked as depreciated.

jordens, 2022-03-28:
Yes. Much better solution!

commit 71851dc3 (2022-04-02): Add slope method to ADC and depreciate max_sample

Closes #343
commit 3e8091ce (2022-04-02): Merge #325 #341 #344

325: Add #[must_use] annotations r=richardeoin a=richardeoin

Use of this attribute was [expanded in the standard library](https://blog.rust-lang.org/2022/01/13/Rust-1.58.0.html#more-must_use-in-the-standard-library) in Rust 1.58, and there is now a [clippy lint](https://rust-lang.github.io/rust-clippy/master/index.html#return_self_not_must_use[](https://github.com/richardeoin)).

Silence the lint for REC methods, since dropping the REC is intended.
Also silence the lint for GPIO modifiers (set_speed) since these are also often dropped.

341: Update sdio-host r=richardeoi [... cut]
PR #344 (2022-04-02): Add slope method to ADC and depreciate max_sample
Closes #343
closed 2022-04-02 by bors[bot]