stm32-rs/stm32h7xx-hal #352: Ethernet/RTIC example for Nucleo-H745ZI-Q board

kelleyk, 2022-04-12:
Hello!

While bringing up the Nucleo-H745ZI-Q that I'm using for a new project, I tried to use the `ethernet-rtic-stm32h747i-disco.rs` example provided in this repository.  I was confused for a little while because, while I could receive frames just fine, outbound frames disappeared as soon as they were handed off to the Ethernet DMA engine.

I eventually had a lightbulb moment and realized that, while the Nucleo-H745ZI-Q and STM32H747I-DISCO are similar in many ways, they might not have exactly the same pin assignments; and sure enough, they differ by exactly one pin:   RMII TXD1 is PB13 rather than PG12.  Once I change this in the example, it works perfectly, without any other changes.

The BSP crate that this repository's README links to does [get this right](https://github.com/antoinevg/nucleo-h7xx/blob/1e3aa9bae284da845c5f7a76c2ce192cf91d91ef/src/pins.rs#L145), but that crate hasn't been updated for the two most recent versions of the HAL, so I figured that I would reach out.

Is this something that folks think might be reasonable to address here, or is better handled in a BSP crate?  (I'm leaning towards the latter, since it's a board-layout-specific issue; but if nothing else, I wanted to open an issue here in case anyone else wanders into the same trouble in the future.)

richardeoin, 2022-04-19:
Thanks for reporting this! It does seem with these two development boards it's an easy mistake to make. To avoid others encountering the same issue I'll add some more documentation at the top of the example.

The best solution is up-to-date and detailed BSPs though. So far in the Rust Community quite often everyone has different stm32h7 boards (based on what was available at the time) and thus there's not a critical mass for a single board. I'd suggest opening a PR on the BSP itself with some updates, especially if you're able to test them out on hardware.

richardeoin, 2022-08-03:
Sufficiently solved by updating the docs, and the [BSP](https://github.com/antoinevg/nucleo-h7xx) has also been updated recently 🎉

closed 2022-08-03 by richardeoin