stm32-rs/stm32h7xx-hal #360: SDMMC no longer working for >= 0.12.0

ostenning, 2022-04-21:
Upon upgrading to `>= 0.12.0` the `sdmmc` implementation fails with Timeout errors. 
I downgraded to SHA `8a74bb4e950166550ea4d3d23e8835d48c55b221` and the problems have been fixed.

cc. @richardeoin

richardeoin, 2022-04-21:
@ostenning [The example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/sdmmc.rs) works fine for me on v0.12. Tested on a STM32H747I-DISCO development board with a SanDisk Extreme 32 GB SDHC UHS-I card.

```
INFO  examples/sdmmc.rs:144 - ----------------------
INFO  examples/sdmmc.rs:145 -
INFO  examples/sdmmc.rs:161 - Read 10 blocks at 24076.043 bytes/s
INFO  examples/sdmmc.rs:162 -
INFO  examples/sdmmc.rs:170 - Wrote block 0
INFO  examples/sdmmc.rs:170 - Wrote block 1
INFO  examples/sdmmc.rs:170 - Wrote block 2
INFO  examples/sdmmc.rs:170 - Wrote block 3
INFO  examples/sdmmc.rs:170 - Wrote block 4
INFO  examples/sdmmc.rs:170 - Wrote block 5
INFO  examples/sdmmc.rs:170 - Wrote block 6
INFO  examples/sdmmc.rs:170 - Wrote block 7
INFO  examples/sdmmc.rs:170 - Wrote block 8
INFO  examples/sdmmc.rs:170 - Wrote block 9
INFO  examples/sdmmc.rs:182 - Done!
```

The recent refactor #341 changed the status check at the end of `write_block` [to use CMD13](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/sdmmc.rs#L722). That matches the F4 HAL, which has been well tested. (Previously this HAL used ACMD13). That might be a clue, but unless you have more details or a solution I have to assume the issue is your setup.

richardeoin, 2022-04-21:
Ok I can reproduce by making multiple writes in succession, without a delay or read in between. There's a fix in #361 , and I also extended the example to cover this behaviour.

ostenning, 2022-04-22:
Good to hear you were able to reproduce it @richardeoin, thanks for the fix ☺️

richardeoin, 2022-04-24:
Closed by #361

closed 2022-04-24 by richardeoin