stm32-rs/stm32h7xx-hal #366: ADC fails to initialize for lower HCLK speeds

ostenning, 2022-06-01:
In an attempt to save power within my application, I've been reducing the `HCLK` value.
I've noticed that when the `HCLK` is set to 100MHz or lower the ADC initialization procedure can sporadically fail. It doesn't happen all the time, and if it doesnt happen, my application runs fine otherwise.

The error encountered is:

```
panicked at 'Cannot start calibration when the ADC is in deeppowerdown-mode', .../src/adc.rs:882:1
```

Investigating this I see the following for the initialization process.

1. Power up the adc, followed by a small 10us delay
2. Begin the calibration process, and check to see that the appropriate flags are set which represent that the adc is powered up within the `check_calibration_conditions` function
3. `check_calibration_conditions` panics because the adc is reported by hardware as being not powered up

I think the issue is that when `HCLK` is set to lower values that this 10us is not large enough for the hardware to set the appropriate flags before the checks happen.

I propose that this could be fixed with a some kind of while loop and timeout mechanism, allowing for devices with lower configured clock speeds.

Any thoughts or feedback is greatly appreciated! 

Kind regards

richardeoin, 2022-09-01:
According to the reference manual this delay is fixed at 10µs, and there's no dependance on the HCLK period. It is a good hypothesis though 

From DS12100, 10µs is the maximum
[image]

richardeoin, 2022-09-01:
@ostenning Would you be able to take a look and see if the patch in https://github.com/stm32-rs/stm32h7xx-hal/pull/386 solves the problem?

closed 2023-11-20 by ostenning