stm32-rs/stm32h7xx-hal #367: STM32H735 Octospi1 hyperbus flash example?

tronical, 2022-06-01:
We're working with an STM32H735 discovery board and were very glad to discover the existing example that shows how to use the 16MB hyperram connected via the octospi2 peripheral.

Connected via the the octospi1 peripheral is the 64MB Macronix MX25LM51245GXDI00 flash storage, for which the octospi.rs example appears to show how to run commands on the controller.

Since the board appears to support accessing that storage also via memory mapping using the hyperbus protocol, we've been trying to make use of that. Unfortunately we haven't had any luck accessing the storage for reading. Short of a GitHub discussion forum, we were wondering if anyone here had any success with that and would be willing to contribute an example?

What we've tried so far is a Hyperbus config based on what I could find in [page 39 in application note AN5050](https://www.st.com/resource/en/application_note/an5050-octospi-interface-on-stm32-microcontrollers-stmicroelectronics.pdf) with `device_size_bytes(26)`, `chip_select_high(3)` and `refresh_interval(0)`.

Unfortunately accessing the memory for reading at `0x90000000` as returned by `init()` doesn't work ("crash").

richardeoin, 2022-06-14:
Hi! Great that you're investigating this. Unfortunately I don't have an example to contribute, but it would be great if someone else can!

As far as I can tell, Macronix MX25LM51245GXDI00 isn't a HyperBus device but rather just an "ordinary" octo-SPI flash (hence the N/A area in [AN5050](https://www.st.com/resource/en/application_note/an5050-octospi-interface-on-stm32-microcontrollers-stmicroelectronics.pdf) Table 5). Have you been able to successfully read from it using Indirect Read mode? I would try that first before moving onto the memory-mapped mode.

tronical, 2022-06-14:
You’re absolutely right. Thanks for spotting, the hyperbus makes no sense here. I’m indeed interested in enabling memory-mapped mode.

I’ll play a bit more with indirect mode and reading/writing.