stm32-rs/stm32h7xx-hal #373: Can no longer write to flash

ostenning, 2022-07-27:
Upon upgrading to the HEAD [commit](https://github.com/stm32-rs/stm32h7xx-hal/commit/6698c1f423ef40d6eb8697b774ad923858d23e8a), my bootloader application can no longer write to the flash. This issue seems to be introduced after tag `0.12.2`.

@richardeoin  @lachlansneff-parallel I see there were some changes made, can you please double check that they are okay?

Kind regards,
Oliver

lachlansneff-parallel, 2022-07-27:
Almost certain this was my fault. I've also been having issues. #371 fixes this for certain though!

richardeoin, 2022-07-27:
@ostenning Thanks for reporting! Could you test with the latest HEAD now that https://github.com/stm32-rs/stm32h7xx-hal/pull/371 is merged please?

ostenning, 2022-08-12:
@richardeoin @lachlansneff-parallel  Good news, from my testing it looks like things are working nicely. I've had my bootloader looping continually reflashing the stm32 for the past few hours and its been working fine

closed 2022-08-12 by ostenning