stm32-rs/stm32h7xx-hal #374: Changing the ADC clock prescaler needs a workaround.

nkrackow, 2022-07-28:
Hi, I've been trying to set the clock prescaler for the internal ADCs and found it to be rather cumbersome.

The [adc initialization function](https://github.com/stm32-rs/stm32h7xx-hal/blob/71e2b75ceaa33e654271144af444c5a4c7dc85b7/src/adc.rs#L461) does not use or do anything with the ADC_COMMON registers in which the clock configuration takes place. Manually changing anything in these registers only works after the ADC is initialized (since it is reset in init) and before it is enabled ([condition form datasheet](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=1042)). 
But then the ADC has to be re-calibrated again since the clock impacts the offset values.
I've tested this [here](https://github.com/quartiq/stabilizer/blob/c040135dde7fcd7e3b6b108ed8a4193ea0016602/src/hardware/driver/adc_internal.rs#L57-L60).

All in all it might be nicer to incorporate the clock prescaler and clock mode from ADC_COMMON into the init function.
But I'm not sure how to do this in the best way. I think it would be a breaking change in any way, right?

nkrackow, 2022-07-28:
Additionally we now found that the 100 MHz maximum clock input that the [HAL checks for](https://github.com/stm32-rs/stm32h7xx-hal/blob/71e2b75ceaa33e654271144af444c5a4c7dc85b7/src/adc.rs#L36) is never a valid clock frequency.
https://github.com/quartiq/stabilizer/issues/571

jordens, 2022-08-09:
"never" as in "in the recent RMs the ADC max clock is 80 MHz".

richardeoin, 2022-08-14:
Thanks for reporting! Indeed the clock prescaler (and potentially also clock mode) should be integrated into the init function. It is a breaking change, but necessary so let's do it.

richardeoin, 2022-09-26:
Closed in #379

closed 2022-09-26 by richardeoin