stm32-rs/stm32h7xx-hal #382: Flash inconsistency error caused by incorrect write procedure

ostenning, 2022-08-18:
I've noticed that in certain circumstances the flash controller returns `BankError::PROGRAMMING_SEQUENCE`. 

It doesn't seem to happen in all circumstances, but the following doesn't work for me within my application:

```rust
let mut buffer: [u8; 256] = [0xFF; 256];
// ... fill buffer with data
flash.erase_sector(Bank::UserBank2, 7).unwrap();
flash
    .write_sector(Bank::UserBank2, 7, 0, &buffer)
    .unwrap();
```

I believe the underlying error is INCERR, inconsistency error. I've created a patch for this.

cc. @richardeoin  @lachlansneff-parallel 

Kind regards

ostenning, 2022-08-18:
@richardeoin , please see https://github.com/stm32-rs/stm32h7xx-hal/pull/383

richardeoin, 2022-08-29:
Closed by #383

closed 2022-08-29 by richardeoin