stm32-rs/stm32h7xx-hal #384: Using ADC with the DMA to continuously read data needs an example

tolgaparlan, 2022-08-29:
For example the [stm32f1xx-hal has it ](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/adc-dma-rx.rs), but it doesn't work with this Hal because `device_peripherals.DMA1` does not expose a split function. 

On the other hand if one tries to follow the other examples under this Hal such as i2c4_bdma.rs
[Go to file](https://github.com/stm32-rs/stm32h7xx-hal/find/master), it doesn't work because the `Transfer::init` function cannot be called with an `Adc` type as it does not implement the `TargetAddress` trait. 

It is not clear at all how to implement this very common use case. Or if it is implemented at all with this crate.

richardeoin, 2022-08-29:
Thanks for reporting @tolgaparlan. That functionality is indeed missing from this crate, since no-one has contributed it yet. You're right that it's a common use case and would be great if we could support it!

This HAL does have equivalent function to `split()`, which [is this line](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/dma.rs#L77). It's doing the same operation of splitting the DMA peripheral into multiple channels/streams. However as you say the missing point is implementing the `TargetAddress` trait for the ADC. I think that should be possible, and would make the DMA API for the ADC similar to other peripherals.

richardeoin, 2022-09-26:
Closed in #389 which added [this example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/adc_dma.rs)

closed 2022-09-26 by richardeoin