stm32-rs/stm32h7xx-hal #393: USB example not working for stm32H743

jg2562, 2022-10-04:
The `usb_serial` example does not work when uploaded to a `Nucleo-H743ZI2`. Also, when testing making my own test, I could never get the USB to initialize and display on my Linux computer. However, when attempted with a `Nucleo-H723ZG` using the `stm32h735` flag, the example worked great and displayed on the computer. So I'm assuming this has to do with how the HAL does the USB configuration or pinouts. Does anyone know why this might be the case?

richardeoin, 2022-10-04:
Thanks for reporting! I took a look at the design of the Nucleo-H743ZI2 board, unfortunately it doesn't quite match the defaults in the example. You'll need to adapt [the usb_serial example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/usb_serial.rs) for it to work. 

The Nucleo-H743ZI2 uses pins PA11 and PA12 for the USB connector CN13. The example [currently uses pins PB14 and PB15](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/usb_serial.rs#L47), so you will need to change it to use pins PA11 and PA12. Perhaps we could make PA11 and PA12 the default in the example?

From what I can tell the Nucleo-H743ZI2 does not use the USB regulator, but in case your board is modified (or a different version) you could investigate enabling it [here](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/usb_serial.rs#L37).

jg2562, 2022-10-04:
Hi, thanks for checking that out! I thought I had tested the different pins of PB14 & PB15 but turns out ig i didn't cause it worked! Does not use the USB regulator, though, as that makes it not appear. It also needs to use USB2 instead of USB1 for the to use the PB pins. As a side note, do you know if you can use a USB though the ST-LINK with device reenumeration?

richardeoin, 2022-10-05:
Great, I appreciate the update. I'll add some documentation to the examples to make the situation clearer.

I don't think it's possible to connect to the H7 USB peripheral through the ST-LINK on a Nucleo board. It might be possible to connect the ST-LINK Virtual Com Port (VCP) to a USART? That would allow communications with the H7 over the ST-LINK USB

commit bfd4044b (2022-10-05): Document USB peripheral allocation in USB examples

Closes #393
commit 7c361157 (2022-10-05): Merge #394

394: Document USB peripheral allocation in USB examples r=richardeoin a=richardeoin

Closes #393
PR #394 (2022-10-05): Document USB peripheral allocation in USB examples
Closes #393
closed 2022-10-05 by bors[bot]