stm32-rs/stm32h7xx-hal #395: SPI over I2S

ostenning, 2022-10-09:
Hello, I'm wondering if its possible to run SPI over I2S?

For example, I have the `SPI2` peripheral with pins `PA9` for SCK and `PC1` for MOSI. Within the Cube HAL I know its possible to run the SPI peripheral as if it were I2S, which would simplify circular DMA implementations.

Is this currently possible?

maxekman, 2023-03-16:
I would also be interested to know about this. As I understand it the I2S peripherals available in the SPIs are separate from the I2S functionality from the SAI peripheral (which can also do more advanced formats). It would be really useful to use the simpler SPI based I2S when not in need of the extended SAI functionality.

liarokapisv, 2024-05-27:
This is definitely possible, the SPI peripheral has a dedicated I2S configuration register that takes over the peripheral. It does not seem to be currently supported by the crate.